ARM-software/CMSIS-DSP #3: aarch64 support

christophe0606, 2022-06-29:
Some build issues reported for cortex-a76 / cortex-a55:

`qsub` intrinsics not supported. But the CMSIS Core_A/Include has not compilation flag to disable the definition of intrinsics like `qsub`.

In `arm_max_f32` and `arm_min_f32`:

```C
/home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/StatisticsFunctions/arm_max_f32.c:168:24: warning: conversion from ‘long unsigned int’ to ‘uint32_t’ {aka ‘unsigned int’} changes value from ‘18446744073709551615’ to
4294967295’ [-Woverflow]
  168 |   maxIdx = vdupq_n_u32(ULONG_MAX);
```

ULONG_MAX is 32 bits so I don't understand why the value is 64 bits according to the compiler.

ua1arn, 2022-09-14:
ULONG_MAX in both places need to replace with UINT32_MAX

christophe0606, 2022-09-15:
@ua1arn Thanks. Already done in a recent update :-)

christophe0606, 2025-04-17:
Lot of Neon has been added to CMSIS-DSP (FFT, Matrix multiply ...).
More will be added.

I close this issue since more support for aarch64 is going to be a long process and continuous improvement.

closed 2025-04-17 by christophe0606