ARM-software/CMSIS-DSP #31: Why the Q15 type of matrix multiplication is transposed, while its Q31 type is not transposed ？

JoleenSun, 2022-07-18:
1. I'm curious about why a transposition is carried out before multiplication in the function arm_mat_mult_fast_q15 , while there is no transposition in the function arm_mat_mult_fast_q31. But transposition can also bring better performance for function arm_mat_mult_fast_q31 actually.
2. Function arm_mat_mult_q15 also has an input "pState", but also has not do the transposition.
Would you please give an explanation？Thanks a lot!

christophe0606, 2022-07-19:
@JoleenSun  Thank you for reopening the issue here.

Transposition is useful when we can use the DSP extensions and instructions like : `__SMLAD`.

But there is no instruction like that for q31. It is probably why the q31 version is not using transposition.

With Helium extension, we indeed have new instructions and transposition may be useful for q31. But, we would need a temporary buffer and there is none in the API. And we need to be backward compatible.

`arm_mat_mult_fast_q15`  is doing the transposition but not by calling the `arm_mat_trans_q15`. Transposition code has been inlined at the beginning of the function.

JoleenSun, 2022-07-19:
Thank you for your answer.
Well, any plan to add new api to allow temp buffer for q31 ?

christophe0606, 2022-07-19:
@JoleenSun Which Cortex-M core are you considering for this optimization ?

I am asking the question because I do think it may be useful for M55 and M85. I am less sure for other cores (but only a benchmark could confirm this).

JoleenSun, 2022-07-19:
Actually,  I haven't really started testing yet.  I just intended to study some interested algorithm implementations currently, during which I encountered some confusions(so I'm asking you for help). 
Thank you for answering…

closed 2022-07-27 by christophe0606