ARM-software/CMSIS-DSP #33: Pre-compiled CMSIS-DSP

pentagon0024, 2022-07-26:
Hi, sorry if it is the wrong place to ask this kind of question

I have a project built on Make, with included pre-compiled math library `libarm_cortexM4lf_math.a`. Then I started moving to the CMSIS-5, as I can play with it in python (many thanks for that!). But I found that it is not easy to build this library and can't find any compiled version. In case I will build it by myself I see some potential problems, such another gcc version that has not been tested or wrong compiler flags.

Could you suggest any solution for that? Is there any place where I can find already compiled version or some reliable way to compile it by myself?

Thank you!

jasonrandrews, 2022-07-26:
I second this request, as a developer working on Linux it would be a big help. If release of a library is not possible, improved instructions to build a library would be helpful.

christophe0606, 2022-07-27:
@jasonrandrews @pentagon0024 Thanks for asking.

We no more release pre-compiled versions because there were too many versions of the library to maintain and validate.

The README on the main page is giving detailed instructions for different way to build: https://github.com/ARM-software/CMSIS-DSP

If it is not enough then tell me what's missing or not clear and I'll improve.

I am covering cmake, CMSIS build and CMSIS Pack to build.

I am not (yet) covering Make.

For Make it is really simple, in each CMSIS-DSP folder you'll see a file without any datatype suffix. Let's take `BasicMathFunctions` as example, you'll see a file : `BasicMathFunctions.c`

Just use this file : it is including all other .c from the `BasicMath` folder.

So, in your makefile you need only a very small number of files and the `Include`  and `PrivateInclude` folder.

Since there is now a dependency to CMSIS Core (as explained in the README), you now also need to add to the includes the `Core/Include` folder from `CMSIS_5` project.

That's all. Then for building it is as usual : ensuring hardware float are used, an option like -O3 or -Ofast. With gcc, avoiding some options which are preventing important optimizations (like -fno-builtins).

If you really want to revalidate your build, the easiest is to use CMSIS build to build the test framework. I'll need to improve documentation on this part if people really want to run the test framework. But I don't think you should have any problem even if you don't run the test framework again.

christophe0606, 2022-07-27:
I have added an example Makefile in folder Source.

pentagon0024, 2022-07-27:
@christophe0606 Thank you for so quick response!

Thank you very much for the explanation, Makefile is extremely helpful - it is clear for me now all the dependencies to build a library.

jasonrandrews, 2022-07-27:
@christophe0606 Christophe, great work, much clearer now.