ARM-software/CMSIS-DSP #38: The function arm_biquad_cascade_df2T_f32 with ARM_MATH_NEON enabled does not work properly

Silfurion, 2022-08-10:
Hello, there seems to be an issue with arm_biquad_cascade_df2T_f32 when stage is bigger than 4 and ARM_MATH_NEON is enabled. In that case the function returns wrong results.

christophe0606, 2022-08-29:
@Silfurion Hello,

Do you have a reproducer ? The filter coefficients and an input signal would be helpful.

You compared the scalar version and Neon version to detect the issue ?

I have tests with more than 3 stages and none is showing any issue.

Silfurion, 2022-08-30:
Hello,

I was able to solve the problem. I had forgotten to use the function arm_biquad_cascade_df2T_compute_coefs_f32 (present in the file arm_biquad_cascade_df2T_init_f32.c) before initializing the biquad. I had trouble finding this function in the documentation. 

I apologize for the inconvenience and wish you a good day.

closed 2022-08-31 by christophe0606