ARM-software/CMSIS-DSP #43: [REQUEST] Add Singular Value Decomposition

dsm, 2022-09-02:
Hi, 
if it is possible, can someone add SVD transform .

dsm, 2022-09-02:
Yes great to hear that, I need non-square matrices for pseudoinverse using SVD, I have 100-200 sensor data for curve fitting as 5 or 6 degree polynomial

dsm, 2022-09-02:
if anything need, I can help

dsm, 2022-09-13:
I think maybe implement svd using method of Golub and Reinsch, that repo is implement in pure c 

[SVD_golub_reinsch](https://github.com/DanielMartensson/CControl/blob/master/src/CControl/Sources/LinearAlgebra/svd_golub_reinsch.c)

Can you review this, please?

christophe0606, 2024-07-08:
A new experimental  C++ extension has been introduced that should make it much easier to write linear algebra algorithms and get correct performances.
It currently does not yet support complex numbers but otherwise should be usable for algorithm like SVD.

We prefer to focus efforts on improving this C++ API and let the ecosystem write the linear algebra algorithms they need.

closed 2024-07-08 by christophe0606