ARM-software/CMSIS-DSP #45: tables/function placement macros/options

ehughes, 2022-09-13:
On Cortex M7 targets such the i.MXRT10xx family,   it is important to be able to control memory placement of the functions/tables for best performance.    For example,   the default placement of functions and twiddle tables into the QSPI flash region.     IN previous versions of the CMSIS DSP,  I would keep a local copy of the source in my repo and "patch" functions and tables like this:

```
FAST_DATA_RAM const q15_t ALIGN4 realCoefAQ15[8192] = {
    (q15_t)0x4000, (q15_t)0xc000, (q15_t)0x3ff3, (q15_t)0xc000, (q15_t)0x3fe7, (q15_t)0xc000, (q15_t)0x3fda, (q15_t)0xc000,
 .
.
.

```

FAST_DATA_RAM would be defined elsewhere so I easily place the table in DTCM, etc.

It would be beneficial for the library to have some mechanism for specifying placement of the routines & tables.    I am migrating to the latest version of CMSIS DSP with the cmake build flow and would like to minimize how much manually modify the library.

christophe0606, 2022-09-14:
@ehughes It is a very good idea. I'll definitely add such a feature.

christophe0606, 2024-06-28:
Done in commit https://github.com/ARM-software/CMSIS-DSP/commit/69d4b8ed8d3271facb2daff94216f7981774e816

closed 2024-07-09 by christophe0606