ARM-software/CMSIS-DSP #48: issue when implementing IIR filter designed by scipy

yf13, 2022-09-24:
Hi,

I am trying to reproduce an IIR filter designed with scipy using CMSIS DSP library on a cortex-M7 device.

Here comes my filter design in python environment:

```
In [66]: b,a=signal.iirfilter(1,[0.01126, 0.03050])
In [67]: b,a
Out[67]:
(array([ 0.02934421, 0. , -0.02934421]),
array([ 1. , -1.93802145, 0.94131158]))
In [68]: signal.lfilter(b,a,[1,2])
Out[68]: array([0.02934421, 0.11555813])
```

Here is my implementation using CMSIS DSP, note that I've negated the a[1:] coeffs below:

```
#define IIR_NUMSTAGES 1
#define IIR_ORDER 2
float32_t _iir_states[IIR_ORDER]={0};
float32_t _iir_coeffs[5*IIR_NUMSTAGES] = { 0.02934421, 0. , -0.02934421, 1.93802145, -0.94131158};
arm_biquad_cascade_df2T_instance_f32 _iir = {IIR_NUMSTAGES, _iir_coeffs, _iir_states};
float32_t x[] = {1, 2};
float32_t exp_y[] = {0.02934421, 0.11555813};
float32_t y[ARRAY_LEN(x)];
arm_biquad_cascade_df2T_f32(&amp;_iir, x, y, ARRAY_LEN(x));
```

But the C implementation outputs `{0.0293442104, -0.00465790788}` on the cortex-m7 MCU.
This result is different from the scipy output.
Since I can't

Regards,
yf13

yf13, 2022-09-24:
Sorry that I just noticed that my biquad instance creation line is having incorrect order of data fields. 
It is likely my usage bug. let me close this ticket first and reopen when necessary.

yf13, 2022-09-25:
I've confirmed my usage with CMSIS-DSP Python wrapper and the result is passed.

christophe0606, 2022-09-26:
@yf13 So you confirm there is no problems ?
I generally convert the filter into second order sections in scipy and then use those for the CMSIS-DSP biquad.

For instance : https://developer.arm.com/documentation/102463/0100/How-to-implement-biquads-to-filter-an-Electrocardiography-signal

Your IIR being already a second order section, what you do look right.

yf13, 2022-10-05:
@christophe0606 , 
Yes I confirmed that CMSIS-DSP library is in line with scipy. 
I only need to use below definition to fix my above fragment:
```arm_biquad_cascade_df2T_instance_f32  iir = {IIR_NUMSTAGES, __iir_states, _iir_coeffs}```

Regards,
yf13

closed 2022-10-05 by yf13