ARM-software/CMSIS-DSP #49: Linear interpolation: incorrect for x outside of the table boundary

jmerkt, 2022-09-27:
https://github.com/ARM-software/CMSIS-DSP/blob/b8177102d9a4aaf83fd3f067364ecfa3100966c2/Source/InterpolationFunctions/arm_linear_interp_f32.c#L91
For values, where `((x - S->x1) / xSpacing)` lies within `(-1.f, 0.f)`, the cast to `int32_t` results in `0` and therefore `i < 0` does not apply. Hence, values outside of the table boundary are not clipped, but wrongly interpolated.
The following code fixes it for be, but results in a slight performance overhead for the float32_t comparison, instead of the original int32_t:
```
/* Calculation of index */
const float32_t x_diff = x - S->x1;
i                      = (int32_t)(x_diff / xSpacing);

if (x_diff < 0.f)
```

christophe0606, 2022-09-28:
@jmerkt Thanks for the bug report !

christophe0606, 2022-09-28:
It should be solved in latest commit.

closed 2022-11-02 by christophe0606