ARM-software/CMSIS-DSP #50: arm_vsi.h is missing

tdjastrzebski, 2022-09-27:
This library does not compile. `arm_vsi.h` is missing

christophe0606, 2022-09-28:
@tdjastrzebski What are you using to build ? 

arm_vsi.h is not part of the library and only used for some examples which are not part of CMSIS-DSP.

startup_ARMCM3.c is is also not part of the library. It used to be delivered as part of the test framework and to run on our IPSS tool. But it has been removed because it is not the goal of the library to maintain boot code. So now the test framework is relying on CMSIS build tools.

So I don't now where you get those 2 files from ?

You should not have to use any of those files to build the library. README is explaining how to do it. You have several choices CMSIS-Build tools, cmake or Makefile.

tdjastrzebski, 2022-09-28:
I simply build all the files one by one with GCC and it does not build.
If `arm_vsi.h` is not part of the library and examples are neither, why those files are referenced in this repo?
References of missing headers are easy to find:  
https://github.com/ARM-software/CMSIS-DSP/search?q=arm_vsi.h  
https://github.com/ARM-software/CMSIS-DSP/search?q=ARMCM3.h  
The best libraries are 'build tool agnostic' because they are meant to be part of other, often much larger systems, using own build tools and because build tools evolve.

christophe0606, 2022-09-28:
The library is build tool agnostic but since it is part of CMSIS ecosystem it is easier to use with CMSIS tools. But we do support Makefile and cmake.

Normally, you should compile only the Source folder as explained in documentation.

The streaming nodes from the ComputeGraph folder are a different project which is dependent on CMSIS-DSP but is not required to use CMSIS-DSP. 

You do not need this ComputeGraph for CMSIS-DSP.

The ARMCM3.h is part of the CMSIS Library. It is only needed if you want to run one of the example on a real device in bare metal (Cortex-M3 in this example)

The files in cmsis_build/RTE folder are needed only if you want to build the example with CMSIS build tools and they use CMSIS pack technology for the boot code for real HW.

In summary : if you follow the README you should be able to build the library ...

The other projects like the examples and tests are relying on CMSIS technology for the boot code.

Some part of the compute graph are made to be used in some Arm Virtual Hardware (AVH) demo and require AVH. But, with compute graph only GenericNodes is required. You use the nodes you need (I repeat you don't need this for CMSIS-DSP).

Since we have more and more people coming from outside of the CMSIS ecosystem, this may be confusing. So, I'll improve the README to highlight what is the core of CMSIS-DSP and build tool and HW independent, and what are the other projects in this repository.

tdjastrzebski, 2022-09-28:
Thank you. I guess, indeed, I got confused. I just included this library using `git submodule add` and I was hoping for immediate success. Not this time ;)

tdjastrzebski, 2022-09-28:
I am showing that in order to build one needs **Source**, **Include** and **PrivateInclude** folders.  
But it still does not build because [FilteringFunctions.c](https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Source/FilteringFunctions/FilteringFunctions.c) and [FilteringFunctionsF16.c](https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Source/FilteringFunctions/FilteringFunctionsF16.c) and similar include C files, rather than headers. That, of course, results in `multiple definition of` errors when linking.
After removing those files I was able to build and link correctly.  
Maybe it would make sense to mention that in README in case one doe snot want to or cannot use Make/CMake.

christophe0606, 2022-09-28:
Indeed, if people don't look at the Makefile content they may not see this problem.
For instance, for Basic Maths, Makefile is just using : BasicMathFunctions.c and all other files are included by this file.

I'll add it to the documentation.

(Reason why it is done like that : on some embedded compilers it makes the compilation much much faster).

tdjastrzebski, 2022-09-28:
I would consider adding `#ifdef` directives around such non-quite-standard C file includes.

christophe0606, 2022-09-29:
I have updated the documentation. I'll consider for a future release what's the best way to handle those not standard C includes (and perhaps remove them).

closed 2022-09-29 by christophe0606