ARM-software/CMSIS-DSP #51: Jinja2 dependency

elagil, 2022-09-28:
Using cmsisdsp as part of a signal processing framework, I stumbled across the limitation of Jinja2 versions. Since 2021, there is a new major release (Jinja2 3.x.y), introducing a few new features and fixes, but cmsisdsp only allows versions <= 2.

Since cmsisdsp is a dependency of mine, I cannot use the latest release of Jinja2. Is there a reason for not allowing the latest release?

Thanks in advance

christophe0606, 2022-09-28:
@elagil Good question.

It is to enable cmsisdsp Python wrapper on Google colab and to enable it to work with some ML packages which are still using Jinja2

I've just checked again and it is still the version 2.11.3 of Jinja in google colab.

It is annoying. 

Unfortunately, the template discovery has changed between 2.0 and 3.0 and it is too much validation work to try to support both versions and add some checks into the code.

I am also annoyed. Because some python packages are using version 2, some others are using version 3 of Jinja.

elagil, 2022-09-30:
@christophe0606  Thanks for the explanation!

So the problem is further up in the dependency chain. I will close this then.

closed 2022-09-30 by elagil