ARM-software/CMSIS-DSP #52: Update the header prefixes to be architecture agnostic

yperess, 2022-09-30:
I have a need for a similar math library in Zephyr which needs to work for and ideally be optimized for additional architectures beyond ARM. I was wondering if it would be possible to rename the headers and replace the `arm_` prefix with `cmsis_`? the `arm_` prefixed headers can still exist and simply include their generic counterparts. This would allow us to provide Zephyr implementations for non arm optimized math logic.

Reference https://github.com/zephyrproject-rtos/zephyr/pull/50012

christophe0606, 2022-10-03:
@yperess I am always trying to help but then it is a problem of priorities. I have a lot in my backlog and not a lot of time. So clearly such a change would have low priority compared to other tasks I have to do.

Also, I need to maintain (as much as possible), backward compatibility. As consequence, I'll never change the name of the functions.

> the arm_ prefixed headers can still exist and simply include their generic counterparts.

Why not doing the opposite ? You could have a `cmsis_dsp_math.h` including the `arm_math.h `

You could also define some macro like 

`#define cmsis_dsp_add_f32  arm_add_f32`

allowing your developers to use `cmsis_dsp_*` functions.

Like that, it would be totally transparent to CMSIS-DSP and I think it would not make your life much more difficult when pulling new CMSIS-DSP version.

Regarding other comments in the Zephyr discussion regarding strongly typed libraries: In pure C, it will be difficult to be as good as C++ for this.

But there are definitely things to improve and I have already a github issues about it. There is lot of legacy in the library and things to improve.

yperess, 2022-10-03:
I'll be happy to send a PR that matches what you described

microbuilder, 2022-10-05:
A `cmsis_dsp_*` wrapper seems sensible to me, and still maintains an association with Arm as the main project maintainer (via `cmsis`), since they deserve credit for their work, without implying that this only works on Arm.

closed 2024-07-09 by christophe0606