ARM-software/CMSIS-DSP #56: Why is `dt` not used in the PID computation?

g-berthiaume, 2022-10-17:
While looking at the [PID Controller algorithm](https://arm-software.github.io/CMSIS-DSP/latest/group__PID.html), I was confused by the omission of the time differential (`dt`) in the gains A0, A1, A2.

The algorithm used by the CMSIS library is 
```
y[n] = y[n-1] + A0 * x[n] + A1 * x[n-1] + A2 * x[n-2]

A0 = Kp + Ki + Kd
A1 = (-Kp ) - (2 * Kd )
A2 = Kd
```

But, to my understanding, it should be 

```
y[n] = y[n-1] + A0 * x[n] + A1 * x[n-1] + A2 * x[n-2]

A0 = Kp + Ki*dt + Kd/dt  //< notice dt
A1 = -Kp - 2*Kd/dt       //< notice dt
A2 = Kd/dt               //< notice dt
````

What is the underlying mathematical assumption here?
Is `dt` assume to always be `1` ?

Regards,

christophe0606, 2022-10-18:
@g-berthiaume Sampling period is only useful to make a correspondence between the continuous time version of your PID and the discrete time version.

If you remain only in discrete time, you don't really need a sampling period. But of course, the coefficients will be different since at the end your discrete time system must work with real sampled signals.

Said differently, you can imagine that dt is part of the definition of the constant.

So, for a given continuous constant K, you'll get different discrete values K' where K' is K/dt or K dt depending on the constant.

But for the CMSIS-DSP implementation, you only care about K'.

Also, the correspondence between continuous and discrete is not only about the sampling time but also how the derivative and integral are approximated.

I don't know what method is used by CMSIS-DSP for integration : forward Euler, backward, trapezoidal ...

I have not been able to derive above formula from Laplace version of the PID.

I'll need to spend more time on it to find what method was used and to improve the documentation.

The discrete PID may be more complex than the one provided by CMSIS-DSP. It depends on how the discrete approximation is computed from the continuous one.

g-berthiaume, 2022-10-18:
Thanks for your detailed answer. I now understand the `K'` silent assumption.
Let me know if you find something. 

Thanks

closed 2022-11-02 by christophe0606