ARM-software/CMSIS-DSP #59: arm_correlate_f32 may write to negative pDst index

stephanosio, 2022-10-20:
arm_correlate_f32 may write to a negative pDst index (before the beginning of the pDst buffer).

This is observed when running the following test cases:

https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Testing/Source/Tests/MISCF32.cpp#L134-L140

https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Testing/Source/Tests/MISCF32.cpp#L142-L148

https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Testing/Source/Tests/MISCF32.cpp#L214-L220

https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Testing/Source/Tests/MISCF32.cpp#L222-L228

https://github.com/ARM-software/CMSIS-DSP/blob/809202bf185280a322efc2e2c850a544747f9d79/Testing/Source/Tests/MISCF32.cpp#L302-L308

christophe0606, 2022-10-27:
@stephanosio The problem is also present on fixed point versions.
Thanks for reporting it.

christophe0606, 2022-10-27:
It should be corrected now.

closed 2022-11-09 by christophe0606