ARM-software/CMSIS-DSP #61: Table config error in RFFT F32 for MVEF?

kjbracey, 2022-10-25:
I've been trying to restrict tables from a Helium build, and hit an error in the conditionalising of the RFFT functions.

My particular case was looking to use `arm_rfft_fast_init_f32(1024)`.

This conditionalises on `ARM_TABLE_BITREVIDX_FLT_512`, but it uses `arm_cfft_init_f32` which conditionalises on `ARM_TABLE_BITREVIDX_FXT_512`.

So both need to be enabled at present to make the RFFT work, although only the `FXT` is used, and armlink can eliminate the `FLT` table anyway.

By the way, the python config helper told me to use the `FLT` form despite selecting Helium. I can see the code has logic to say `FXT` instead, but it didn't seem to activate for me. Some confusion about `HELIUM` versus `config["HELIUM"]` perhaps? And it should perhaps be switching on (MVEF && !AUTOVECTORISE) anyway.

christophe0606, 2022-10-27:
@kjbracey The logic for this table selection has gone too complex and I need to rethink it to have something easier to maintain.
Sorry you have found and issue here. I am pretty sure there are other problems and inconsistencies.

I had opened an issue to track this enhancement : https://github.com/ARM-software/CMSIS-DSP/issues/6

I need to have something that can be used with cmake, make and the MDK configuration wizard (not yet used in CMSIS-DSP).

kjbracey, 2022-10-27:
Maybe you can combo that with the enabling control of the memory section control to make life easier, getting the compiler+linker to do the work, reducing or totally avoiding the need for config in the first place.

If your memory section control gives a prefix, then you put each table in a separate section - effectively forcing `-fdata-sections` for them. That then ensures the linker can eliminate unreferenced tables, as it did the FLT table for me here. And it gives the user the option of fine-grained control in the scatter file, although I'm sure most would wildcard.

Then you don't need all the fine-grained knowledge of what code uses what table in initialisation. You just need the ability to configure which sizes appear in the `switch` statements. That's back to being simple enough to not really need a tool any more.

Ideally you'd avoid having `switch` statements for what's effectively a config choice - that design total anathema to embedded devices, because of the way they pull in every possibility, despite the thing being switched on almost certainly being known at compile-time.

I'd suggest exposing `arm_rfft_1024_fast_init_f32` etc as API. Anyone using those APIs then would not have to think about table config at all. Only people insisting on passing length as a variable through the older API would hit the issue.

And even the switch API could work better if it was made `inline`, giving the compiler a chance to compile-time reduce the switch. Hopefully it would be bright enough to not actually inline it if it didn't know the value of `len`.  Still, shouldn't rely on the inline and simplification occurring to get the saving - you don't want your debug builds to have 100K more tables than non-debug. So you'd still want to have the direct-size-specifying API.

kjbracey, 2022-10-27:
Another thought, although it's getting a bit complex. (But if it eliminates config/make complexity...?)

Have an option to use weak references/alias in the switch statement, so you don't get a table unless you've requested it elsewhere in the code with a `arm_rfft_1024_fast_enable_f32()` call, that does nothing except serves to pull it in as a non-weak reference.

A powerful pattern, but can be annoying to get right in a portable way.

christophe0606, 2022-10-27:
@kjbracey Thanks for good suggestions.

I have made an update. Since the table used in scalar and MVE versions of `cfft` are different, the #if needed to test both cases.

I have not modified the Python script because at some point I'll remove it and use something else ... so it is still wrongly generating a `_FLT_` in MVE case.

> Ideally you'd avoid having switch statements for what's effectively a config choice - that sort of API design is total anathema to embedded devices, because of the way they pull in every possibility, despite the thing being switched on almost certainly being known at compile-time.

I agree. But the other problem of not using a init function and giving details about the implementation is that we may break user code if we need to do things differently  ...

With the init, it gives us more flexibility to change things internally without having to modify the exposed public APIs.

But, the drawback is that we reference tables and prevent linker from doing the work.

We have still not converged on the best tradeoff and obviously some improvements are required.

kjbracey, 2022-10-27:
> With the init, it gives us more flexibility to change things internally without having to modify the exposed public APIs.

Makes sense. Although I'd counter that the length of an FFT is almost as fundamental to the caller as the data type - indeed is part of the data type, no? -, so it's not unreasonable for you to break out separate functions. I'd also compare with C++'s pair of `span<T>` and `span<T, len>` and similar type pairs - both are available, depending on need, because the constant-len is so common, and worth optimising.

C++14 has made it more plausible to allow this sort of thing to be done automatically, giving the best of both worlds. You can keep everything looking like it's a run-time parameter, but still have compile-time optimisation as long as they're all constants.

The `constexpr` in C++ is enough to basically force the optimisation when all parameters are constant, even in debug builds. C's not quite as nice, sadly, but still there's a strong chance that a C++14-capable compiler will do the equivalent for a C `inline` with constant parameters if not debug, it's just not guaranteed in the same way.

I'd previously been investigating that sort of model for pinmux-type things - the runtime "pin" parameters to Mbed OS driver objects, allowing pinmux tables to be eliminated if the pin was (as it usually is) constant, without changing API.  Also CRC configuration. We've reached a time where the static-analysis C++ can guarantee via `constexpr` is potentially a significant benefit to embedded programming compared to C.

kjbracey, 2022-10-27:
Not tested, but looks good by inspection. Thanks!

closed 2022-10-27 by kjbracey

christophe0606, 2022-10-27:
@kjbracey  I am indeed thinking about a C++ API for CMSIS-DSP using constexpr and template to avoid the code duplication that we have in the library and to provide a more user friendly API. But, it is long term plan ...