ARM-software/CMSIS-DSP #63: Improve consitency with different compilers

petertorelli, 2022-10-26:
The following two header files should be modified as shown so that when compiling with Arm Clang or  GCC on Ubuntu, the C function macros and the inline intrinsic functions are used. I believe this is correct, unless there is additional setup I missed? In my use-case I compile only the individual DSP function-files that are needed rather than linking the library. I'm building on these two non-Arm platforms as a simple reference-code example that is portable to any OS (EEMBC's AudioMark benchmark) with the intention that an Arm target would use NEON and the enhancements.

`arm_math_types.h`

```C
#elif defined ( __APPLE_CC__ ) || defined ( __GNUC__ )
#include <stdint.h>
#define  __ALIGNED(x) __attribute__((aligned(x)))
#define __STATIC_FORCEINLINE static inline __attribute__((always_inline)) 
#define __STATIC_INLINE static inline
```

`dsp/none.h`

```C
#if defined ( _MSC_VER ) || defined(__GNUC_PYTHON__) || defined(__APPLE_CC__) || defined (__GNUC__)
    __STATIC_FORCEINLINE uint8_t __CLZ(uint32_t data)
    {
      if (data == 0U) { return 32U; }

      uint32_t count = 0U;
      uint32_t mask = 0x80000000U;

      while ((data & mask) == 0U)
      {
        count += 1U;
        mask = mask >> 1U;
      }
      return count;
    }
:
:
```

petertorelli, 2022-10-26:
I'll generate a PR if this is a legit issue.

christophe0606, 2022-10-27:
@petertorelli You can just define `__GNUC_PYTHON__`.
The name of the define is not very clear. It was initially introduced for building the Python wrapper but now it is generally used to build for host.

I don't want to have automatic detection like with __GNUC__ because if you compile with gcc and target Arm you will want some DSP instructions existing on Cortex-A aarch32.

The meaning of `__GNUC_PYTHON__` is to disable those DSP extensions and use instead pure C implementation.

In fact, what I am trying to detect is not the compiler but the host platform. So when it is built with MSVC or XCode I can detect the platform from the compiler. When it is built with gcc or clang I can't. So the developer just has to defined `__GNUC_PYTHON__`

petertorelli, 2022-10-27:
I understand now, thanks. (I've tried to determine platform architecture from predefined compiler macros before and it was maddening due to utter lack of standardization.)

closed 2022-10-27 by petertorelli