ARM-software/CMSIS-DSP #66: MVE integer absmax functions can return -1.0

kjbracey, 2022-10-29:
Looking at the MVE implementation of `arm_absmax_no_idx_q15` (and I assume the others are similar), if the data contains `0x8000`, it returns this as the result. This is not what the other implementations do.

`curExtremValVec` is correctly modelled as unsigned (`VMAXA` produces unsigned results from 0 to 0x8000), but the final gathering uses `vmaxavq` instead of `vmaxvq` on it, with this type mismatch being indicated by the presence of the cast to `q15x8_t`. (Doesn't change the result directly, but contributes to the type confusion).

Then the final `uint16_t` from `vmaxvq` needs to be saturated to get the `q15_t` result from 0 to 0x7fff.

kjbracey, 2022-10-29:
While in there, just noticed that `blkCnt` is 16-bit for the `q15` function, in a probable cut-and-paste error, limiting you to 512K values.

Other notes:

* Using `blkCnt` and having a tail section seems redundant anyway - wouldn't a simple counter and single predicated loop work better? Produces better code on armclang. And it's what the non-`no_idx` do.
* `curExtremeValVec` should be initialised via `vdupq_n_u16`
* `maxValue` would be better `uint16_t`, as it gets the unsigned result from `vmaxvq` then you can finish with `*pResult = __USAT(maxValue, 15)`.

christophe0606, 2022-11-02:
@kjbracey Thanks for good feedback. I'll have a look today.

kjbracey, 2022-11-02:
Just to reassure you - I'm not planning on going through and critiquing all the implementations randomly!

I did happen to see this in my debug output, so ended up staring at it. (On audio data I'd just expanded to be full-range - I noticed the scale accuracy problem while doing that).

christophe0606, 2022-11-02:
@kjbracey No worry. It is good feedback and helps improve the library.

christophe0606, 2022-11-03:
It should be corrected now.

kjbracey, 2022-11-03:
Looks good by inspection. Only thought is that `clip_q63_to_q31` is overkill.

Could do it with `*pResult = maxValue - (maxValue >> 31);`

closed 2022-11-03 by kjbracey