ARM-software/CMSIS-DSP #69: Building for AArch64 with LLVM/Clang

cberk1, 2022-11-07:
Hi, 

I hope I'm not overlooking something obvious but I'm trying to build some of the library filtering functions for Android (AOSP, Cortex A55). To my knowledge I've followed the build instructions in README.md but I'm getting errors. I'm not sure whether the errors are a result of compiling for AArch64 or because I'm using Clang. My Blueprint file looks something like this:
`
    srcs: [
        "Source/FilteringFunctions/FilteringFunctions.c"
    ],
    
    cflags: [
        "-DDISABLEFLOAT16",
        "-DARM_MATH_NEON",
        "-D__GNUC_PYTHON__",
        "-Ofast",
        "-mfpu=neon",
        "-mcpu=cortex-a55",
        "-mthumb",
        "--target=arm-none-eabi"
    ],

    local_include_dirs: [
        "PrivateInclude",
        "Include",
        "ComputeLibrary",
        "libcmsis_5/CMSIS/Core/Include"
    ],

    export_include_dirs: [
        "Include",
        "ComputeLibrary",
        "Include/dsp",
        "libcmsis_5/CMSIS/Core/Include"
    ]
    }
`

where one of the multiple errors is this:
`CMSIS/Core/Include/cmsis_gcc.h:299:46: error: value size does not match register size specified by the constraint and modifier [-Werror,-Wasm-operand-widths]
  __ASM ("rev %0, %1" : __CMSIS_GCC_OUT_REG (result) : __CMSIS_GCC_USE_REG (value) );`

Whilst I'm aware of possible incompatibilies with clang ([here](https://github.com/ARM-software/CMSIS_5/issues/1507)), this looks like a 32/64 bit problem. Would anyone be able to point me in the right direction to get this building?

Are there any updates on Clang support?

christophe0606, 2022-11-08:
@cberk1 If you're building for AArch64 you should not use the CMSIS Core. So the line
`libcmsis_5/CMSIS/Core/Include` 

should be removed.

Now, it looks like `cmsis_gcc.h` (from Core) is nevertheless included and it is not normal because with `__GNUC_PYTHON__` it should be bypassed.

If you look at `CMSIS-DSP\Include\arm_math_types.h` you'll see:

```
#elif defined (__GNUC_PYTHON__)
#include <stdint.h>
#define  __ALIGNED(x) __attribute__((aligned(x)))
#define __STATIC_FORCEINLINE static inline __attribute__((always_inline)) 
#define __STATIC_INLINE static inline

#else
#include "cmsis_compiler.h"
#endif
```

So, `cmsis_compiler.h` should not be included. It is this header which is loading definition from CMSIS Core,

The meaning of the error : The compiler files of CMSIS Core are defining some C macros (and functions) to use some Arm instructions which are not available on AArch64. So, they should not be used.

cberk1, 2022-11-08:
@christophe0606 Thank you very much for this - I've stripped out the unnecessary includes and added the `__GNUC_PYTHON__` to ALL my Android makefiles and got past the error.

 I'm now on to a new error:
```
error: implicit declaration of function '__SMLAD' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        acc0 = __SMLAD(x0, c0, acc0);`
```
which is raised by `arm_biquad_cascade_df1_fast_q15.c:195`. As I understand it, this should be defined in `dsp/none.h` and depends on whether the `ARM_MATH_DSP` flag is defined. I can't see where I'm defining `ARM_MATH_DSP`, the only place I can see it being set in the code is at the top of `arm_math_types.h` when `__ARM_FEATURE_DSP` is defined as 1. I'll keep looking.
Any thoughts?

christophe0606, 2022-11-08:
`__ARM_FEATURE_DSP` should not be 1 
If it was 1 then the `cmsis_gcc.h` should not have generated errors.
When it is 1 it means the DSP instructions are available. But they are not on Aarch64.

So, if it is 1 it is a problem. You should check the compilation options.

The specification "Arm C Language extensions" is saying:

> __ARM_FEATURE_DSP is defined to 1 if the DSP (v5E) instructions are supported and the intrinsics defined in
Saturating intrinsics are available. These instructions include QADD, SMULBB and others. This feature also implies
support for the Q flag.
__ARM_FEATURE_DSP and its associated intrinsics are deprecated in ACLE 2.0 for A-profile. They are fully supported
for M and R-profiles. This macro is defined for AArch32 only.

cberk1, 2022-11-08:
Thank you for the reply. 

As it turns out the Android build was building both for 32 and 16 bit - very confusing! Thank you for pointing me in the right direction!

closed 2022-11-09 by cberk1