ARM-software/CMSIS-DSP #71: arm_rfft_fast_f32 causes Bus Fault on M33 for some FFT sizes

cberk1, 2022-11-21:
Hi, 

I'm currently trying to get the real FFT function working on an M33. The function works perfectly for an FFT size of 512 but when increasing the size to 1024 or above a bus fault exception occurs like this:

```
### EXCEPTION ###
PC =0024C194, ExceptionNumber=D'-11, EXC_RETURN=FFFFFFED

R0 =20083D38, R1 =20082D38, R2 =2007ED38, R3 =20081D38
R4 =20080D38, R5 =20084D38, R6 =2007DD3C, R7 =80000008
R8 =00000078, R9 =00008000, R10=00001000, R11=00000078
R12=00000078, SP =20082D00, LR =00001000

MSP=20169FD8, PSP=20082C98
MSPLIM=20168F00, PSPLIM=200822D0

PRIMASK=00, FAULTMASK=00, BASEPRI=40, CONTROL=00
XPSR=21100000, APSR=nzCvq, EPSR=01000000, IPSR=00 (NoException)

ICSR =0440F805, AIRCR=FA050000, SCR  =00000000, CCR  =00000611
SHCSR=00070002, CFSR =00008200, HFSR =00000000, AFSR =00000000
MMFAR=80000004, BFAR =80000004
FaultInfo : (BusFault)
FaultCause: (Precise data bus error) (BFAR valid)

Stack:
  20082CC0: 00000000 80000000 00000000 00000000
  20082CD0: 00000000 80000000 00000000 00000000
  20082CE0: 7FFF0208 FFFF0208 00000000 00000000
  20082CF0: 00000000 80000000 00000000 00000000
* 20082D00: 00000000 00000000 00000000 00000000
  20082D10: 00000000 00000000 00000000 00000000
  20082D20: 00000000 00000000 00000000 00000000
  20082D30: 00000000 80000008 2007DD3C 20084D38
  20082D40: 20080D38 20083D38 20082D38 2007ED38
  20082D50: 20081D38 20087E14 20087A54 00000200
  20082D60: 00000001 20087694 2007D978 2007D97C
  20082D70: 49465853 3F99999A 43480000 439D1462
```

where my code is
```
    int fft_size = FFT_SIZE;

    static float32_t fft_in[FFT_SIZE];
    static float32_t fft_out[FFT_SIZE];

    /*CMSIS FFT initialise*/
    arm_rfft_fast_instance_f32 S;
    arm_rfft_fast_init_f32(&S, fft_size);
    arm_rfft_fast_f32(&S, fft_in, fft_out, 0);
```

Being new to these type of issues, would anyone be kind enough to point me in the right direction to debug this? What are the possible causes?

christophe0606, 2022-11-22:
@cberk1 Can you check the return code of `arm_rfft_fast_init_f32` to see if an error occurred during the initialization.

Do you use any compilation flag to avoid the inclusion of all FFT tables and keep only the ones you need ? (to minimize code size)

cberk1, 2022-11-22:
Hi Christophe, thank you for your reply.

Your are right - the initialization is failing (ARM_MATH_ARGUMENT_ERROR) for 1024, 2046 and 4096 fft sizes.
```
    arm_rfft_fast_instance_f32 S;
    int ret = arm_rfft_fast_init_f32(&S, 4096);
``` 
Here are my compilation flags:
```
ccflags-y += -DARM_TABLE_TWIDDLECOEF_F32_16\
             -DARM_TABLE_BITREVIDX_FLT_16\
             -DARM_TABLE_TWIDDLECOEF_RFFT_F32_32\
             -DARM_TABLE_TWIDDLECOEF_F32_16\
             -DARM_DSP_CONFIG_TABLES\
             -DARM_FFT_ALLOW_TABLES
```

I'm not using any others.

christophe0606, 2022-11-22:
@cberk1 The compilation flags are wrong. You only include tables for RFFT of length 32. You need to change this to include the tables for the RFFT 1024 if you want to compute with this FFT length. It is the cause of the crash (and I am surprised it worked for length 512. Perhaps it did not crash but the result was probably wrong).

cberk1, 2022-11-22:
@christophe0606 this was of course the issue. Given the somewhat obscure memory errors I didn't think to dive into the CMake stuff, thanks for pointing me in the right direction. In the end, un-commenting the desired flags in arm_tables_config.h worked on my platform.

closed 2022-11-22 by cberk1