ARM-software/CMSIS-DSP #72: arm_sqrt_f32 not optimised for ARM Compiler 6

kjbracey, 2022-11-25:
`arm_sqrt_f32` has some intent to allow VSQRT.F32 to be inlined, bypassing `sqrtf`' s need to do `errno` handling.

This is implemented for ARM Compiler 5 and IAR, but not for ARM Compiler 6, so anything with square root is slower than it need be.

ARM Compiler 6 can be simply handled using `_sqrtf()` (single underscore) from `<math.h>`, which is already included - this already has its own conditional for hardware FP or not, so no local test is needed, unlike ARMC5 or IAR.

(Took me a little while to get there - ACLE specifies `__sqrtf()`, but ARMC6 fails to provide this correctly. It's has an implementation in `<math.h>` not `<arm_acle.h>`, and it only appears as `std::__sqrtf()` in C++ mode.  And `__builtin_sqrtf()` seems to just act as `sqrtf()`. But `_sqrtf()` is perfect.)

I haven't investigated the GCC situation - but I can see there's no handling.

christophe0606, 2022-11-28:
@kjbracey Thanks for the report. I'll make the update as soon as I have finished the task I am working on.

kjbracey, 2022-11-28:
Turns out `_sqrtf()`isn't _quite_ perfect, as its current implementation forces the same input and output register on the VSQRT.F32, which ended up causing an unnecessary VMOV.F32 in my example. I filed a ticket for that.

So you could have your own assembler, IAR-style, to improve on the current `_sqrtf()` implementation, but it's probably best to stick with the library, for future-proofing.

I believe use of `-Ofast` or whatever to shut off the errno handling globally would render this moot, but I think it's useful to have the specific sqrt optimisation for people who aren't comfortable doing that.

christophe0606, 2022-11-30:
@kjbracey  I have made the fix and added a case for gcc too. I hope my `#if` are not too wrong for gcc. I tested on M55 and M0 and the tests looked ok.

You'll also see a case `__GNUC_PYTHON__` It is the test I am using when building the Python wrapper on Linux. In this case I won't test for GCC and use this version instead.

kjbracey, 2022-11-30:
Looks good to me.  Only suggestion might be an overall structural simplification to eliminate repeat `else`s for basic `sqrt`.

```
#if defined ( __CC_ARM ) && defined __TARGET_FPU_VFP
      *pOut = __sqrtf(in);

#elif defined ( __ARMCC_VERSION ) && ( __ARMCC_VERSION >= 6010050 )
      *pOut = _sqrtf(in);
      
#elif defined ( __ICCARM__ ) && defined __ARMVFP__
      __ASM("VSQRT.F32 %0,%1" : "=t"(*pOut) : "t"(in));

#elif defined ( __GNUC__ ) && !defined(__GNUC_PYTHON__) && defined (__VFP_FP__) && !defined(__SOFTFP__)
      __ASM("VSQRT.F32 %0,%1" : "=t"(*pOut) : "t"(in));
      
#else
      *pOut = sqrtf(in);
#endif
```

closed 2022-11-30 by kjbracey