ARM-software/CMSIS-DSP #76: Python wrapper cmsis_arm_correlate_f32 wrong output calculation

a1lu, 2022-12-14:
Hi,
the python wrapper calculates the output length like this:
```
uint32_t outputLength = 2*MAX(srcALen,srcBLen) - 1 ;
```
for 
```
int srcALen = 3;
int srcBLen = 8
```
The result is 8. Thats wrong.
On my side a fix is to wrap the MAX macro with braces

```
uint32_t outputLength = 2*(MAX(srcALen,srcBLen)) - 1 ;
```

a1lu, 2022-12-14:
```
#define MAX(A, B) ((A) < (B) ? (B) : (A))
```

in `PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_module.h` would fix this everywhere

christophe0606, 2022-12-15:
@a1lu Thanks for feedback. I have merged the PR.
I don't update the binary versions of the wrapper too often but a new version will probably come in january.

PR #77 (2022-12-15): Add braces around MAX macro
The current implementation leads to wrong calculation of output lengths.

Fixes #76
closed 2022-12-15 by christophe0606