ARM-software/CMSIS-DSP #88: Install CMSIS-DSP to VisualStudio + VisualGDB for STM32

NeiD52, 2023-02-06:
Hello, I need a step-by-step instruction on how to add the SMSIS-DSP library to the VisualStudio + VisualGDB development environment. I'm having a lot of trouble with the integration, after adding the main library functions I can't compile the project. Compilation errors appear **undefined reference to `function'**.  I read the forum on VisualGDB.com, but there is not complete information, it is not very reliable. Please help me how to do it step by step? Thank you in advance.

christophe0606, 2023-02-06:
@NeiD52 Hello. What is your target ? Do you want to compile for Cortex-M ?

NeiD52, 2023-02-06:
> @NeiD52 Hello. What is your target ? Do you want to compile for Cortex-M ?

Yes, I want to compile for Cortex-M3 - stm32f103c8t6? And I don't use Keil and STM32CubeIDE. I noticed a lot of differences in the directories of the latest library v1.14.3. I noticed a lot of differences in the directories of the latest library. I wanted to use the previous version 1.4.7, but there is no way to mark arm_atan2_f32.

christophe0606, 2023-02-06:
I have never done it from Visual. But if you include the files BasicMaths.c and similar ones from each Source sub-folder it should be enough.
Define also -DDISABLEFLOAT16 because float16 on M3 is useless.

Also use the Include and PrivateInclude folders.

The section of the README about building with Makefile may be helpful too since it is going to be very similar regarding the files to build and include to define.

christophe0606, 2023-02-06:
@NeiD52 It is a global settings for the compiler command line. So, there should be a place in Visual where you define those kind of settings.

We can't document all IDEs. The supported way to build is through CMSIS build tools and CMSIS Packs. Although we also support cmake and document Makefile.

For all other way of building, you have to adapt the explanations.

jaguerra767, 2023-02-22:
Related question, I'm trying to build using the makefile but I keep getting the error that "there is no rule to make target ...CMSIS-DSP/Source/BasicMathFunctions/BasicMathFunctions.c.o', needed by `builddir/libCMSISDSP.a'.  Stop" (Redacted my path)

closed 2023-02-27 by christophe0606