ARM-software/CMSIS-DSP #90: reverse transform rfft in Python wrapping function

ooshyun, 2023-02-15:
I'm testing fixed point fft, ifft, and stft. However, I got an issue with rfft in reverse transform. I tried from fft to ifft, but the length of final output is smaller than the original wav file. In this case, I chopped the wavfile as much as the window size. I attached a code below,

The code said the length of the original waveform is 256, but the output from irfft is 254. Is it the proper operation?

```python
import warnings
import numpy as np
import cmsisdsp as dsp 
import librosa
import resampy 
print()
print((librosa.ex("trumpet")))
wav, sr = librosa.load(librosa.ex("trumpet"))
wav = resampy.resample(wav, sr, 16000)
sr = 16000
window_size = 256 # 128, 256, 512, 1024
wav = wav.flatten()
wav = wav[:window_size]

wav_rfft = np.fft.rfft(a=wav, n=window_size)
wav_cfft = np.fft.fft(a=wav, n=window_size)

wav_rfft_reference = np.zeros(shape=(window_size*2+2,), dtype=np.float32)
wav_rfft_reference[:window_size][0::2] = wav_rfft[:-1].real
wav_rfft_reference[window_size+2:][0::2] = np.flip(wav_rfft[:-1].real)
wav_rfft_reference[:window_size][1::2] = wav_rfft[:-1].imag
wav_rfft_reference[window_size+2:][1::2] = -np.flip(wav_rfft[:-1].imag)
wav_rfft_reference[window_size] = wav_rfft[-1].real
wav_rfft_reference[window_size+1] = wav_rfft[-1].imag

wav_rfft_bin = np.abs(wav_rfft)
wav_cfft_bin = np.abs(wav_cfft)

wav_irfft = np.fft.irfft(a=wav_rfft, n=window_size)
wav_icfft = np.fft.ifft(a=wav_cfft, n=window_size)

wav_q15 = dsp.arm_float_to_q15(wav)
wav_q31 = dsp.arm_float_to_q31(wav)

rfft_instance_q15 = dsp.arm_rfft_instance_q15()
status = dsp.arm_rfft_init_q15(rfft_instance_q15, window_size, 0, 1)
rfft_instance_q31 = dsp.arm_rfft_instance_q31()
status = dsp.arm_rfft_init_q31(rfft_instance_q31, window_size, 0, 1)

rfft_1_q15 = dsp.arm_rfft_q15(rfft_instance_q15, wav_q15) 
rfft_1_q31 = dsp.arm_rfft_q31(rfft_instance_q31, wav_q31)

# fft cmsis_dsp real and imag ~ ifft
# rfft
shift = np.log2(window_size)
rfft_instance_q15 = dsp.arm_rfft_instance_q15()
status = dsp.arm_rfft_init_q15(rfft_instance_q15, window_size, 1, 1)
rfft_instance_q31 = dsp.arm_rfft_instance_q31()
status = dsp.arm_rfft_init_q31(rfft_instance_q31, window_size, 1, 1)

irfft_1_q15 = dsp.arm_rfft_q15(rfft_instance_q15, rfft_1_q15) 
irfft_1_q31 = dsp.arm_rfft_q31(rfft_instance_q31, rfft_1_q31)

irfft_1_q15_revert = dsp.arm_q15_to_float(irfft_1_q15)*window_size
irfft_1_q31_revert = dsp.arm_q31_to_float(irfft_1_q31)*window_size

if irfft_1_q15_revert.shape != wav.shape:
    warnings.warn(f"Q15, Q31 wav length is shorter than original")
    print("Error Max in fft Q15 -> ifft Q15: ", np.max(wav[...,:irfft_1_q15_revert.shape[-1]]-irfft_1_q15_revert))
    print("Error Max in fft Q31 -> ifft Q31: ", np.max(wav[...,:irfft_1_q15_revert.shape[-1]]-irfft_1_q31_revert))
else:
    print("Error Max in fft Q15 -> ifft Q15: ", np.max(wav-irfft_1_q15_revert))
    print("Error Max in fft Q31 -> ifft Q31: ", np.max(wav-irfft_1_q31_revert))
```

christophe0606, 2023-02-15:
@ooshyun You're right. There is a bug in the wrapper. I am going to correct it in the github.
The update on PyPI will take more time because I have other fixes to do before and generating the binaries for PyPI is very time consuming (since unfortunately I have not yet found a way to automate it and target all the devices I want to target)

christophe0606, 2023-02-21:
It is corrected (I hope :-) ) in latest commit https://github.com/ARM-software/CMSIS-DSP/commit/8b49478b1ff7bae3f7b44f14c87f87ecf86de88c

I will upload it to PyPI at some point but it may take some time because I have changed how the binaries of the wrapper are built and I need more testing before I upload something.

I have updated the PythonWrapper README to explain how to built from source.

ooshyun, 2023-02-21:
I'll test, and let you know the results!🥹

ooshyun, 2023-02-22:
I tested rfft in reverse transformer, and the result is **perfect!** I attached the test code and result. 

```python
"""
irfft return smaller size of fft frame

Issue report: https://github.com/ARM-software/CMSIS-DSP/issues/90#issuecomment-1431311949
"""

import warnings
import numpy as np
import cmsisdsp as dsp 
import librosa
import resampy 
print()
print((librosa.ex("trumpet")))
wav, sr = librosa.load(librosa.ex("trumpet"))
wav = resampy.resample(wav, sr, 16000)
sr = 16000
window_size = 256 # 128, 256, 512, 1024
wav = wav.flatten()
wav = wav[:window_size]

wav_rfft = np.fft.rfft(a=wav, n=window_size)
wav_cfft = np.fft.fft(a=wav, n=window_size)

wav_rfft_reference = np.zeros(shape=(window_size*2+2,), dtype=np.float32)
wav_rfft_reference[:window_size][0::2] = wav_rfft[:-1].real
wav_rfft_reference[window_size+2:][0::2] = np.flip(wav_rfft[:-1].real)
wav_rfft_reference[:window_size][1::2] = wav_rfft[:-1].imag
wav_rfft_reference[window_size+2:][1::2] = -np.flip(wav_rfft[:-1].imag)
wav_rfft_reference[window_size] = wav_rfft[-1].real
wav_rfft_reference[window_size+1] = wav_rfft[-1].imag

wav_rfft_bin = np.abs(wav_rfft)
wav_cfft_bin = np.abs(wav_cfft)

wav_irfft = np.fft.irfft(a=wav_rfft, n=window_size)
wav_icfft = np.fft.ifft(a=wav_cfft, n=window_size)

wav_q15 = dsp.arm_float_to_q15(wav)
wav_q31 = dsp.arm_float_to_q31(wav)

rfft_instance_q15 = dsp.arm_rfft_instance_q15()
status = dsp.arm_rfft_init_q15(rfft_instance_q15, window_size, 0, 1)
rfft_instance_q31 = dsp.arm_rfft_instance_q31()
status = dsp.arm_rfft_init_q31(rfft_instance_q31, window_size, 0, 1)

rfft_1_q15 = dsp.arm_rfft_q15(rfft_instance_q15, wav_q15) 
rfft_1_q31 = dsp.arm_rfft_q31(rfft_instance_q31, wav_q31)

# fft cmsis_dsp real and imag ~ ifft
# rfft
shift = np.log2(window_size)
rfft_instance_q15 = dsp.arm_rfft_instance_q15()
status = dsp.arm_rfft_init_q15(rfft_instance_q15, window_size, 1, 1)
rfft_instance_q31 = dsp.arm_rfft_instance_q31()
status = dsp.arm_rfft_init_q31(rfft_instance_q31, window_size, 1, 1)

irfft_1_q15 = dsp.arm_rfft_q15(rfft_instance_q15, rfft_1_q15) 
irfft_1_q31 = dsp.arm_rfft_q31(rfft_instance_q31, rfft_1_q31)

irfft_1_q15_revert = dsp.arm_q15_to_float(irfft_1_q15)*window_size
irfft_1_q31_revert = dsp.arm_q31_to_float(irfft_1_q31)*window_size

assert irfft_1_q15_revert.shape == wav.shape

print("Error Max in fft Q15 -> ifft Q15: ", np.max(np.abs(wav-irfft_1_q15_revert)))
print("Error Max in fft Q31 -> ifft Q31: ", np.max(np.abs(wav-irfft_1_q31_revert)))
```

Thanks!

christophe0606, 2023-02-22:
@ooshyun Thank you for the feedback ! It confirms my own tests. So, I'll upload binaries to PyPI.

closed 2023-02-22 by christophe0606