ARM-software/CMSIS-DSP #91: FFT init function teke  so much memmory

Hoseinbme, 2023-02-17:
I wanted to get a Fourier transform on the stm32 MCU With the help of cmsis functions

But I encountered the problem of lack of memory while the buffer size was small.
even when set the length of FFT 32 .
After a lot of trouble, by checking the map file, I realized that a function consumes a lot of memory when called, about 150 KB. arm_rfft_fast_init_f32()

I wanted to get a Fourier transform on the stm32 MCU With the help of cmsis functions

But I encountered the problem of lack of memory while the buffer size was small.

After a lot of trouble, by checking the map file, I realized that a function consumes a lot of memory when called, about 150 KB. arm_rfft_fast_init_f32()

Do you have any suggestions to reduce the memory size? This function sets the initial conditions, but it requires a lot of memory.
Now, I am not in a position to quickly provide hardware with more memory.

[tester for fft.log](https://github.com/ARM-software/CMSIS-DSP/files/10765096/tester.for.fft.log)

HERE IS THE MAP FILE   SERCH FOR   arm_common_tables.o      AND YOU SEE PARTS THAT CONSUME MEMMORY

christophe0606, 2023-02-20:
@Hoseinbme By default, all FFTs tables are included.
You need to define some flags for the compiler so that useless FFTs tables are not included. Like that you can tune the size of the library for what you want.

For instance, those flags should enable all interpolation tables (like used from computing sine) and only the tables for RFFT FAST F32. Float16 is also disabled since you probably don't need it:

`-DARM_DSP_CONFIG_TABLES 
-DARM_FAST_ALLOW_TABLES 
-DARM_FFT_ALLOW_TABLES
-DARM_TABLE_TWIDDLECOEF_F32_16 
-DARM_TABLE_BITREVIDX_FLT_16
-DARM_TABLE_TWIDDLECOEF_RFFT_F32_32 
-DARM_ALL_FAST_TABLES
-DDISABLEFLOAT16`

You can use Python to generate those flags:
`pip install streamlit`

and from CMSIS-DSP folder:

`streamlit run cmsisdspconfig.py`

Hoseinbme, 2023-02-25:
@christophe0606    I did this, but the problem is still there, did I make a mistake?
```
---------------------------------------------------*/

#define ARM_DSP_CONFIG_TABLES
#define ARM_TABLE_TWIDDLECOEF_F32_16            
#define ARM_TABLE_BITREVIDX_FLT_16
#define ARM_TABLE_TWIDDLECOEF_RFFT_F32_32
#define ARM_FFT_ALLOW_TABLES

#include "main.h"
#include <math.h>
#include "arm_math.h"
#include <stdio.h>
.
.
.
 ``
   `

christophe0606, 2023-02-27:
@Hoseinbme Those options must be defined at the global level when you compile all of CMSIS-DSP.
So you need to pass those options on the compiler command line :
`-DARM_DSP_CONFIG_TABLES -DARM_TABLE_TWIDDLECOEF_F32_16 ...`

Otherwise those options are not seen when the CMSIS-DSP is compiled and the tables are still included.

Depending on how you build the library, there are different way to do this.
In general in an IDE you'll have a way to define this in the project options.

closed 2023-03-14 by christophe0606