ARM-software/CMSIS-DSP #93: const-qualify table of Y values in interpolation

escherstair, 2023-02-27:
Interpolation functions use Instance structures as this one
https://github.com/ARM-software/CMSIS-DSP/blob/d7e4dea51ad19724adcd2c9f82eb42af12e0f5fc/Include/dsp/interpolation_functions.h#L50-L59

The `pYData` field points to the table of Y values, that must not (and is not) modified by the interpolation function.
As done for other functions, I suggest to const-qualify the field, changing to
```
          const float32_t *pYData;          /**< pointer to the table of Y values */
```
And so on, wherever is necessary.

christophe0606, 2023-02-28:
@escherstair Thank you for the feedback. I'll do the change as soon as I have finished the tasks I am working on currently.

christophe0606, 2023-03-09:
Done in latest commit.

closed 2023-04-03 by christophe0606