ARM-software/CMSIS-DSP #99: Global macro MASK_INDEX in Include\dsp\utils.h is poorly named for global namespace.

cliffordslocombe, 2023-03-30:
The rather generic name in global scope is prone to symbol clash.

In my case it caused a error with an identically names private member constant in a C++ class.  As a _private_ member the name is not unreasonable as there is no risk of a clash, but macros third-party library headers having no means to narrow scope scope or namespace need to be more carefully named.

Users should not have to change their reasonable naming and scoping to account for CMSIS's poor choices.

```
#define INDEX_MASK         0x0000003F
```

Equally perhaps for `SQ` and `ROUND_UP` in the same file.  Since they are presumably only required to support the inline `arm_recip_q31()` in the same header, the simple solution would be to undefine them after use at the end of the header:

```
#undef INDEX_MASK
#undef SQ
#undef ROUND_UP
```

christophe0606, 2023-04-03:
@cliffordslocombe Thanks for feedback. I'll improve this.

spe-ciellt, 2023-10-24:
Hello!

This causes problem in using CMSIS-DSP with Zephyr. With the release of Zephyr 3.5 it is a separate module that needs to be included. There is a major name clash in the macros `ROUND_UP()` and `ROUND_DOWN()`, which is already used by Zephyr.

Why not tell Zephyr to change their name then? I tried and got a reference to their coding guideline. https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-3-macro-name-collisions

My current "fix" is to change the order of the include files since CMSIS DSP at least check if the macro is defined before defining a new macro. Zephyr doesn't.

christophe0606, 2023-10-25:
@spe-ciellt It should be corrected in latest commit.
I also corrected other naming issues (SQ, INDEX_MASK ...)

closed 2023-11-27 by christophe0606