ARM-software/CMSIS-DSP #104: no support for Cortex-R5F processors

edbek, 2023-05-18:
It is strange that there is no support for Cortex-R5F processors. For example, for new TI AM243x processors.

christophe0606, 2023-05-22:
@edbek Unfortunately, for DSP extensions, the library is dependent on CMSIS-Core.
And CMSIS-Core is still essentially Cortex-M oriented.

CMSIS-DSP only needs instructions macros (like QADD etc ...) and compiler abstraction from CMSIS-Core. But CMSIS-Core has to provide much more.

If at some point the DSP intrinsics and compiler abstractions were moved outside of CMSIS-Core, then it should be easy to support Cortex-R.

You may open an issue on the [CMSIS](https://github.com/ARM-software/CMSIS_5) project regarding CMSIS-Core refactoring.

christophe0606, 2023-05-22:
Issue closed since problem is on CMSIS-Core side.

closed 2023-05-23 by christophe0606