ARM-software/CMSIS-DSP #109: Mistake in function arm_split_rfft_q31 ?

JoleenSun, 2023-06-21:
[image]

Codes in line 278 and line 288 are inconsistent with there comments. 

Comment: pBTable  Code: CoefA2

Would you please tell me the whether it's a mistake with the code or the comment.

christophe0606, 2023-07-04:
@JoleenSun It is a documentation issue

JoleenSun, 2023-07-05:
OK  thanks for reply. With all due respect, it seems like an algorithm issue more for only real part of coefB has been related in the current calculation which both the image and real part of the coefB are related in arm_split_rfft_q15.
Maybe I don't  understand the lgorithm well enough for the above words. Looking forward to your reply.
Thanks.

christophe0606, 2023-07-05:
It is confusing but I think the code is right.

In Q31 version you have `multSub_32x32_keep32_R (outR, *pIn2, CoefA2)` and it is a **sub** 

But Q15 version is using `outR = (outR + (*(pSrc2 + 1) * *(pCoefB + 1))) >> 16;` and it is a **add**

But now if you look at the definition on the tables, you see that:

`pATable[2 * i + 1] = 0.5 * (-1.0 * cos (2 * PI / (double) (2 * n) * (double) i));`

and 

`pBTable[2 * i + 1] = 0.5 * ( 1.0 * cos (2 * PI / (double) (2 * n) * (double) i));`

They are opposite values.

Q31 code looks correct : it has replaced an **add** with a **sub** and at same time used a value of the B table instead of the A table.

Note that it only works for odd values of the table.

`pBTable[2*i]` is not opposite to `pATable[2*i]`

JoleenSun, 2023-07-05:
Wow,  I rederived  the codes related according to your explain. The codes are really  no problem. Thank you so much.

closed 2023-07-05 by JoleenSun