ARM-software/CMSIS-DSP #110: In-place filtering, complex computation

Alouettesu, 2023-06-26:
Can input and output arrays in filtering functions and complex math functions be same?

```
arm_fir_f32(&S, samples, samples, N);
arm_biquad_cascade_df2T_f32(&S, samples, samples, N);
arm_cmplx_mag_f32(pSrc, pDst, numSamples);
```

If this is possible, it should be mentioned in documentation. If impossible, these arguments must be marked with `restrict` attribute.

JustasSiauliai, 2023-09-15:
Hi Cristopher, did you have any chance to review this issue?

christophe0606, 2023-09-15:
@JustasSiauliai Unfortunately, documenting all of this is a change that will take a long time since there are lots of function to review and update.
The most secure is to assume you can't do in-place filtering.