ARM-software/CMSIS-DSP #114: Some problem while adding file path to the cmake

suguguan, 2023-08-03:
**Issue Report: Trouble with Including CMSIS-DSP Source Files in CLion**

**Develop Environment:** CLion

**Problem Description:**
I followed the documentation to add the following commands to my CMakeLists.txt file:

```cmake
set(CMSISCORE "CMSIS-DSP/Source")
add_subdirectory(CMSIS-DSP/Source bin_dsp)
```

After reloading CMake and compiling, I encountered the first issue of missing `arm_math.h`. To address this, I included the paths `CMSIS-DSP/Include` and `CMSIS-DSP/PrivateInclude` using `include_directories()`.

Upon compiling again, I encountered a new error: `main.c:(.text.startup.main+0x14): undefined reference to 'arm_cos_q31'`. To resolve this, I added the line `file(GLOB_RECURSE SOURCES "CMSIS-DSP/Source/*.*")` to the CMakeLists.txt.

However, this led to yet another issue with multiple definitions. After inspecting the files, I discovered that each sub-file in the Source directory included the others, causing the conflicts. As a temporary solution, I manually deleted specific files like `CMSIS-DSP/Source/WindowFunctions/WindowFunctions.c`, which allowed the project to compile successfully.

While this approach worked, it requires deleting multiple files, which is cumbersome and unsatisfactory. I also attempted to remove the `file(GLOB_RECURSE SOURCES "CMSIS-DSP/Source/*.*")` by using `list(REMOVE ITEM ...)` and `foreach()`, but it was unsuccessful and time-consuming.

**Request for Assistance:**
I would like to seek guidance on how to properly include the CMSIS-DSP source files in my project without facing the issues of missing headers and multiple definitions. If the problem lies with my approach, I would appreciate advice on the correct method to resolve this situation. Additionally, if feasible, I propose considering the removal of conflicting files like `WindowFunctions.c` to streamline the usage of the CMSIS-DSP library.

christophe0606, 2023-08-03:
@suguguan CMSIS Core is not from CMSIS-DSP but from the https://github.com/ARM-software/CMSIS_5 project.
You need to point to the folder Core.

Something like : set(CMSISCORE CMSIS_5/CMSIS/Core)

You'll need also to link the CMSIS-DSP library by adding something like:

target_link_libraries(my_application PUBLIC CMSISDSP)

Then you can include the arm_math.h from your project.

suguguan, 2023-08-03:
> @suguguan CMSIS Core is not from CMSIS-DSP but from the https://github.com/ARM-software/CMSIS_5 project. You need to point to the folder Core.
>
> Something like : set(CMSISCORE CMSIS_5/CMSIS/Core) [...]

"Thank you very, very much for your advice. I have understood your point, and I don't need to make many modifications to compile successfully, which is exactly what I wanted. Let me summarize: CMSIS-DSP is part of CMSIS_5, and it still requires some related files from CMSIS_5. Therefore, we need to specify the location of CMSIS_5, which corresponds to the 'Drivers/CMSIS' path generated by CubeMX. As a result, we should set the value of the 'CMSISCORE' variable to the 'Drivers/CMSIS' path, not the 'Include' or 'Source' folders of CMSIS-DSP or any other folders."

closed 2023-08-03 by suguguan

Mexico-zyy-lost, 2024-04-18:
> Thank you very, very much for your advice. I have understood your point, and I don't need to make many modifications to compile successfully, which is exactly what I wanted. Let me summarize: [...]

could u share ur last cmakelists.txt file?