ARM-software/CMSIS-DSP #116: DARM_DSP_CONFIG_TABLES

hosseinb1994, 2023-08-21:
Hello. 
I am working on a project that already DARM_DSP_CONFIG_TABLES used inside the cmake file. But I got this error: Severity	Code	Description	Project	File	Line	Suppression State Error	G3B2E063B	#error ("-DARM_DSP_CONFIG_TABLES no more supported. Use the new initialization functions to let the linker optimize the code size.")	C:\Users\hosse\sphereprojectbase\SphereProjectBase_RT\SphereProjectBase_RT	C:\Users\hosse\sphereprojectbase\SphereProjectBase_RT\lib\CMSIS_5\CMSIS\DSP\Include\arm_math_types.h	277	 Would you please let me know how to solve the error?

christophe0606, 2023-08-30:
@hosseinb1994 You cannot use any more the flag ARM_DSP_CONFIG_TABLES and related ones. It was getting too complex to use and maintain.

Instead now we are relying on the linker to do the code size optimization.
But it may requires a few changes in the code.

For instance, if you use CFFTs and know their size at build time, don't use the `arm_cfft_init` but instead prefer one of the new init functions like `arm_cfft_init_512_f32` for a CFFT on 512 samples.

Same for RFFT, MFCC and other datatypes.

Of course, it will help to build with `-ffunction-sections` and  `-fdata-sections` and use the right linker flags to do the optimization (compiler dependent).

Now what surprises me is this : `SphereProjectBase_RT\lib\CMSIS_5\CMSIS\DSP\Include\arm_math_types.h`

Because this change to the compilation flag is part of the latest CMSIS-DSP release and it is no more part of CMSIS_5. CMSIS-DSP is now a separate project.

So if you use the CMSIS-DSP in CMSIS_5 you should not get this error message.

hosseinb1994, 2023-09-14:
Thank you for your reply. I will replace them and make change on code. Then let you know about result.

hosseinb1994, 2023-09-14:
Thank you for your reply. Yes you are right, I am using CMSIS-DSP inside CMSIS_5 and get this error. So basically can I add arm_cfft_init_512_f32 and other c files like this inside the source file? Or do I need to do other action?

hosseinb1994, 2023-09-14:
Thank you for your reply. Yes you are right, I am using CMSIS-DSP inside CMSIS_5 and get this error. So basically can I add arm_cfft_init_512_f32 and other c files like this inside the source file? Or do I need to do other action? And where can I find arm_cfft_init_512_f32 and other c file like this. Because I used arm_dot_prod_f32.c, arm_abs_f32.c and .... inside my cmakefile.

hosseinb1994, 2023-09-14:
I also checked CMSIS-DSP from a link in your profile in github. But I could not find this function: arm_cfft_init_512_f32 . Could you please let me know where can I find them? Thank you in advance

christophe0606, 2023-09-14:
It is defined through macros : https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/TransformFunctions/arm_cfft_init_f32.c#L107

It is easier to see in the header : https://github.com/ARM-software/CMSIS-DSP/blob/main/Include/dsp/transform_functions.h#L231

You can either use CMSIS-DSP from this github repository with cmake or make or use the CMSIS-Pack.
CMSIS-Pack may already be supported by your IDE otherwise you can use command line tools for CMSIS build tools.

If you have the error but don't have the new FFT init functions
then there is a mix between the new and old CMSIS-DSP in your environment.

In CMSIS_5 you don't have the compilation check and the ARM_DSP_CONFIG_TABLE is supported.

hosseinb1994, 2023-09-19:
Thank you for your nice answer.  Two more question: How can I use cmsis-pack and If I have the error but don't have the new FFT init functions how can I figure out this issue? Thank you in advnace

christophe0606, 2023-09-19:
@hosseinb1994 If you have the error but not the new FFT then you need to use a CMSIS-DSP package that is not coming from your environment.

Either getting the version from this github or CMSIS Pack.

CMSIS Pack are generally provided by the IDE but may not be available in all IDEs yet.

If they are not available in your IDE, you can use the command line tools : https://github.com/Open-CMSIS-Pack/cmsis-toolbox

CMSIS toolbox and pack are a pack manager and a build system.
But it may be difficult to only use the CMSIS-DSP Pack. Generally the pack cover everything in your build : boot code, driver etc ...

closed 2024-02-08 by christophe0606