ARM-software/CMSIS-DSP #122: package build one single problem

tdjastrzebski, 2023-10-28:
It is a great package with one single yet significant issue.
For no apparent reason it is completely dependent on one particular build tool (CMAKE) and requires significant effort to compile with anything else.

christophe0606, 2023-11-02:
@tdjastrzebski The README is explaining how to build with a Makefile.
It is also explaining how to adapt to other build tools : https://github.com/ARM-software/CMSIS-DSP#how-to-build-with-any-other-build-system

and it is also supporting CMSIS build tools.

tdjastrzebski, 2023-11-02:
@christophe0606 Thank you, I admit that I missed that.
It may be a good idea to instead of removing/excluding `xxFunctions.c` files, put their content in (e.g.) blocks like this
```
#if defined(COMPILE_xxFUNCTIONS_FILES_ONLY)
//
#endif
```
or
```
#if !defined(COMPILE_ALL)
//
#endif
```

closed 2023-11-16 by christophe0606