ARM-software/CMSIS-DSP #124: buffer overflow in arm_conv_q15 and q7

christophe0606, 2023-11-02:
Running tests on Zephyr for mps3_an547 (Cortex-M55 with DSP support)
west build -p -b mps3_an547 tests/lib/cmsis_dsp/filtering -T libraries.cmsis_dsp.filtering.misc.fpu -t run
A buffer overflow is detected.

Code in the tested functions seem suspicious: [arm_conv_q15](https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FilteringFunctions/arm_conv_q15.c#L195) and [arm_conv_q7](https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FilteringFunctions/arm_conv_q7.c#L189).

The loop in [arm_conv_q31](https://github.com/ARM-software/CMSIS-DSP/blob/00bab39e8f64ca32aa6707ea5e1fa6ca95dd5276/Source/FilteringFunctions/arm_conv_q31.c#L195) does not display the same suspicious exit condition.

For what it’s worth, changing those conditions to either >= 2 or > 1 solves the issue and lets the test pass successfully.

christophe0606, 2023-11-24:
Tentative fix in the latest commit.

ithinuel, 2023-11-24:
This can be confirmed by running the Zephyr tests for cmsis-dsp (eg [here](https://github.com/zephyrproject-rtos/zephyr/blob/f89c5ddd1aaf0e4a5b6fd0a1397c5a72f1cba388/tests/lib/cmsis_dsp/filtering/src/misc_q15.c#L77) on an547 with:

`west build -p -b mps3_an547 tests/lib/cmsis_dsp/filtering -T libraries.cmsis_dsp.filtering.misc.fpu -t run`

christophe0606, 2023-11-24:
It was confirmed as solved (private communication).

closed 2023-11-24 by christophe0606