ARM-software/CMSIS-DSP #125: zero-crossing detection with CMSIS-DSP?

tdjastrzebski, 2023-11-02:
Does CMSIS-DSP implement any algorithm which can be used for zero-crossing detection (ZDC)?
I understand that it is a broad question and zero-crossing detection is a science in itself, but I will appreciate any suggestions on whether or not CMSIS-DSP is the right tool for such task.

christophe0606, 2023-11-03:
@tdjastrzebski There is currently no such algorithm. But the CMSIS-DSP kernels can be used to build some.

To compute the ZCR itself you could multiply the signal with a delayed version by 1 sample. 
You may use CMSIS-DSP to remove any DC bias from the signal.

As an example of use of the CMSIS-DSP Python wrapper I had created this notebook:

https://colab.research.google.com/github/ARM-software/CMSIS-DSP/blob/main/PythonWrapper/examples/kws_example/kws.ipynb

It still use NumPy for the zero crossing counting. But other parts (filtering, classifying etc ...) are done with CMSIS-DSP.
The zero crossing could partially be done with CMSIS-DSP too.

The above example is not a good example of keyword spotting and don't really work. It is just an example to demonstrate a more complex use case and you may find some ideas to reuse in it.

tdjastrzebski, 2023-11-07:
@christophe0606 Thank you for sharing this solution!

closed 2023-11-07 by tdjastrzebski