ARM-software/CMSIS-DSP #137: how to re-build the cmsisdsp  source code to reflect changes in python wrapper as well

mehreenjabeen, 2023-12-17:
Hi,

If I want to add some print statements in [https://github.com/ARM-software/CMSIS-DSP/tree/main/Source/TransformFunctions], what steps do I have to follow to reflect the changes? Which folder do I exactly need to re-build.  I am trying to see the output of the print statement that I added via the python wrapper.

Thanks

christophe0606, 2023-12-18:
@mehreenjabeen I have not rebuilt the Python wrapper since the CMSIS Compute graph has moved to a new repository : CMSIS-Stream and a new python package cmsis-stream.

But it looks like I'll need some work to remove the remaining dependencies in the Python wrapper.

You can try to remove all references to cg (for compute graph);
In the MANIFEST.in, in setup.py.

I think it should solve your problem.

mehreenjabeen, 2023-12-18:
Thanks @christophe0606 .  I tried running ` pip install -e .` but got ```
ERROR: Failed building editable for cmsisdsp
Failed to build cmsisdsp
ERROR: Could not build wheels for cmsisdsp, which is required to install pyproject.toml-based projects. `python setup.py buils` is also rresulting in error  `x86_64-linux-gnu-gcc` -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.8/PythonWrapper/cmsisdsp_pkg/src/cmsisdsp_filtering.o PythonWrapper/build_linux/bin_dsp/libCMSISDSP.a -o build/lib.linux-x86_64-3.8/cmsisdsp_filtering.cpython-38-x86_64-linux-gnu.so
x86_64-linux-gnu-gcc: error: PythonWrapper/build_linux/bin_dsp/libCMSISDSP.a: No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

christophe0606, 2023-12-18:
In your error message I see this:

> error: PythonWrapper/build_linux/bin_dsp/libCMSISDSP.a: No such file or directory 

You need to first build the CMSIS-DSP library. The Python wrapper will then link this library `libCMSISDSP.a`

To build, you must customize the paths in the `create.sh` script in your `build_linux` folder. Once you have succeeded in running this `create.sh`, then you can just type `make`

mehreenjabeen, 2023-12-18:
Thanks @christophe0606 , it worked.

closed 2023-12-18 by mehreenjabeen