ARM-software/CMSIS-DSP #146: Include/arm_math_types_f16.h: undefined __ARM_FEATURE_MVE

matthewbstokes-zz, 2024-01-10:
Missing check that `__ARM_FEATURE_MVE` is defined before checking its value

 `Include/arm_math_types_f16.h:52`
```
#if !(__ARM_FEATURE_MVE & 2)
  #if !defined(DISABLEFLOAT16)
    #if defined(__ARM_FP16_FORMAT_IEEE) || defined(__ARM_FP16_FORMAT_ALTERNATIVE)
      typedef __fp16 float16_t;
      #define ARM_FLOAT16_SUPPORTED
    #endif
  #endif
#else
```

Update to:
```
#if !defined(__ARM_FEATURE_MVE) || !(__ARM_FEATURE_MVE & 2)
```

christophe0606, 2024-01-11:
Corrected in latest commit

closed 2024-01-11 by matthewbstokes-zz

Ivanovitchk, 2025-04-08:
the same issue is also present in arm_math_types.h (the non _f16 version).

Include/arm_math_types_f16.h:333
```
#if !defined(__ICCARM__) || !(__ARM_FEATURE_MVE & 2)
  typedef float float32_t;
#endif
```