ARM-software/CMSIS-DSP #151: Issues with log implementation for Neon

christophe0606, 2024-01-30:
`vlogq_f32` is used directly to implement the `arm_vlog_f32`.
But `vlogq_f32` being based on Taylor expansion for Neon, it can be used only for a limited range of the argument. So `arm_vlog_f32` should not use it directly but apply some scaling when the argument is not in the right range.

Gao-HaoYuan, 2024-01-30:
I've conducted some tests, and when the value is very close to 1, there are significant differences in the calculation results. It might be beneficial to pay attention to values close to 1, such as 1.000000119, 1.00000119, and 0.9999911. 

This critical issue ：
I extracted the code for vlogq_f32 from CMSIS-DSP, When calculating log(1.000000119), the result is -0 instead of +0.

christophe0606, 2024-01-30:
@Gao-HaoYuan I think we should do something like computing log(1.000000119/2) + log(2)

I need to study the Taylor expansion used to find the right bounds for the argument.

Gao-HaoYuan, 2024-01-30:
[Log_test.zip](https://github.com/ARM-software/CMSIS-DSP/files/14093894/Log_test.zip)

This is a code example that requires modification of the IP and NDK addresses.

 I attempted to understand the coefficients of the Taylor expansion in the log, but I couldn't figure out the source of the coefficients in the code. It might be a special value from (x-n) of the Taylor expansion. If I have the time, I will also try to solve this issue.

Gao-HaoYuan, 2024-06-21:
I found a similar solution that can refer to the log calculation logic of OpenCV. However, I haven't organized the code for individual execution yet. If you have time, you can refer to it.  

[mathfuncs](https://github.com/opencv/opencv/blob/4.x/modules/core/src/mathfuncs.cpp)

@christophe0606

christophe0606, 2024-06-21:
@Gao-HaoYuan Thank you. Sorry for being slow at looking at this issue ...

christophe0606, 2024-07-10:
@Gao-HaoYuan I have just tested on architecture "armv8-a+simd"
I don't see any issue.

With your values : `{0.9f, 1.000000119, 1.1f, 1.2f}` 

I am getting:

`-0.105358951    0.000000267     0.095310278     0.182321578`

and the expected values are:

`-0.105361  1.19*10^-7  0.0953102  0.182322`

Gao-HaoYuan, 2024-07-10:
@christophe0606 Maybe I made a mistake. Thank you very much for your effort, and I apologize for taking up your time. I will go check and correct my errors.

christophe0606, 2024-07-10:
@Gao-HaoYuan I have tested with the clang compiler. Maybe it is an issue with the compiler.
I have also tested with -Ofast and it is not and issue of IEEE compliance since -Ofast build is behaving the same.

Gao-HaoYuan, 2024-07-10:
@christophe0606 My compiler is indeed Clang. Thank you very much for helping me discover the issue.

closed 2024-07-11 by christophe0606