ARM-software/CMSIS-DSP #156: Value range of arm_atan2_*

marcluque, 2024-02-15:
I was going over the [atan2 documentation](https://arm-software.github.io/CMSIS-DSP/latest/group__atan2.html), but couldn't find any indication of the radian range that is used. Can somebody help me determine this? I tried going over the code, but it doesn't seem obvious to me what value range can be expected. I'm thinking that it is `(-Pi, Pi]`, but it would be great to have that confirmed.
Would also be great if the documentation for `atan2` could be updated with the expected range.

While going over the [atan2 code](https://github.com/ARM-software/CMSIS-DSP/blob/647b755ad80d53ecca56a555508084663f97c0eb/Source/FastMathFunctions/arm_atan2_q15.c), I noticed that the implementation slightly deviates from an exemplary definition given on [Wikipedia](https://en.wikipedia.org/wiki/Atan2#Definition_and_computation). In particular, [this line](https://github.com/ARM-software/CMSIS-DSP/blob/647b755ad80d53ecca56a555508084663f97c0eb/Source/FastMathFunctions/arm_atan2_q15.c#L204) does not seem to be "correct", since the case is never possible for `q15_t`. It seems like this code might have been taken from the floating point implementation where this distinction makes sense. Another thing hinting at this is [this line](https://github.com/ARM-software/CMSIS-DSP/blob/647b755ad80d53ecca56a555508084663f97c0eb/Source/FastMathFunctions/arm_atan2_q15.c#L4).

christophe0606, 2024-02-19:
@marcluque You're right. This function needs a bit of cleaning.
The tests are comparing it with the Numpy arctan2 so the output should be between -pi and pi.

marcluque, 2024-02-19:
@christophe0606 thank you for checking and giving the confirmation!

christophe0606, 2024-03-06:
Documentation improved.

closed 2024-03-06 by christophe0606