ARM-software/CMSIS-DSP #158: Implementation .c-Files not being copied correctly in STM32CubeMX - missing from .pdsc?

K1ngjulien, 2024-02-22:
Hi!

### Problem

After finally figuring out how to get  the cmsis-dsp package when using STM32CubeMX (manually downloading and installing the .pack for cmsis core and cmsis-dsp from https://www.keil.arm.com/packs/ ) i've run into another roadblock when trying to use it, they used to be avalable by default a few years back)

After getting the packages installed and selected in configurator, i enabled them, clicked "generate code" and tried to compile, with the following result:

```
...
[  2%] Building C object CMakeFiles/flugdatenlogger.elf.dir/Middlewares/Third_Party/ARM_CMSIS/Source/BasicMathFunctions/BasicMathFunctionsF16.c.obj
[...]/flugdatenlogger/Middlewares/Third_Party/ARM_CMSIS/Source/BasicMathFunctions/BasicMathFunctions.c:29:10: fatal error: arm_abs_f32.c: No such file or directory
   29 | #include "arm_abs_f32.c"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[...]/flugdatenlogger/Middlewares/Third_Party/ARM_CMSIS/Source/BasicMathFunctions/BasicMathFunctionsF16.c:29:10: fatal error: arm_abs_f16.c: No such file or directory
   29 | #include "arm_abs_f16.c"
      |          ^~~~~~~~~~~~~~~
...
```

after some investigation, I noticed that the implementation files `arm_abs_f32.c`, `arm_abs_f16.c`, etc. were not being copied into my project directory:

[image]

Only the "root" files, which include the implementation .c files are copied.

I checked the directory where STM32CubeMX installs its packages, but to my surprise, the files where there:

[image]

Out of curiosity, I then added one of those implementation files to the `ARM.CMSIS-DSP.pdsc` file

[image]

after restarting CubeMX (probably required to reload the package description from disk) and clicking "Generate Code" the file appeared!

[image]

### Question

Now to my actual question(s).

Are the implementation .c files missing from the .pdsc  file on purpose? Seems to me like they should be in there

https://github.com/ARM-software/CMSIS-DSP/blob/647b755ad80d53ecca56a555508084663f97c0eb/ARM.CMSIS-DSP.pdsc#L66

If not, would it be possible for you to add all the implementation ".c" files to the package description, so that they will be copied correctly in future versions?

Alternatively, could this be an issue with CubeMX not following the spec for package descriptions correctly? 

Not sure if here's the correct place to report this, but i figured ST would take forever to respond :D

Thanks in advance!
Cheers, K1ngjulien

christophe0606, 2024-02-22:
@K1ngjulien I've discussed with people involved in the definition of pdsc and pack standard. It looks like the problem may be with the STM32CubeMX.

Also the files from the pack should not be copied into the project. They should be referenced from the project directly. So it may explain the issue : the IDE is not copying the `arm_abs_f32.c` but this file should not have to be copied.

jkrech, 2024-02-22:
@K1ngjulien just for me to follow the steps correctly ...
You launch STM32CubeMX, create a new project for a board/device and then generate code for CubeIDE ?
What are the settings in "Code Generator" section?
"Copy only the necessary library files"
or
"Add necessary library files as references in the toolchain project configuration file"

I think the latter will prevent STM32CubeMX from attempting to copy files into your workspace and this way you should be able to avoid you missing files.

Note that if a module is listed as a source file of a component in the pdsc file it will trigger the invocation of the compiler with that file. Since here some c modules are looped in via `#include` they must not be build as individual modules.
Unfortunately STM32CubeMX makes the assumption that the list of source files to be compiled also represents the list of modules that need to be copied if they should reside in project folder. This is not covered by the Open-CMSIS-Pack specification as only files with the config attribute represent configuration files that are application specific and user editable. All other files are "read-only" and should only be referenced.

ReinhardKeil, 2024-02-29:
@iomint it seems to be a problem in the way CubeMX handles `#include` files. Whereas `*.h` files are copied, files with extension `*.c` seem not to be copied.

iomint, 2024-03-01:
I will have a look and discuss it with my colleagues in charge of STM32CubeMX

K1ngjulien, 2024-03-04:
>  just for me to follow the steps correctly ...
You launch STM32CubeMX, create a new project for a board/device and then generate code for CubeIDE ?
What are the settings in "Code Generator" section you are using?

Yes, i was using "copy only the necessary library files". With STM32CubeIDE, I seems to be working when using the library reference.

Unfortunately I'm using CLion, and the generated CMakeLists is now missing the *all* source files, including HAL...

```
home/julian/.local/share/JetBrains/Toolbox/apps/clion/bin/cmake/linux/x64/bin/cmake --build /home/julian/code/dsp-test/cmake-build-debug --target dsp-test.elf -j 6
[8/8] Linking C executable dsp-test.elf
FAILED: dsp-test.elf 
: && /usr/bin/arm-none-eabi-gcc -g -Wl,-gc-sections,--print-memory-usage,-Map=/home/julian/code/dsp-test/cmake-build-debug/dsp-test.map -mcpu=cortex-m4 -mthumb -mthumb-interwork -T /home/julian/code/dsp-test/STM32F411VETX_FLASH.ld CMakeFiles/dsp-test.elf.dir/Core/Src/main.c.obj CMakeFiles/dsp-test.elf.dir/Core/Src/stm32f4xx_hal_msp.c.obj CMakeFiles/dsp-test.elf.dir/Core/Src/stm32f4xx_it.c.obj CMakeFiles/dsp-test.elf.dir/Core/Src/syscalls.c.obj CMakeFiles/dsp-test.elf.dir/Core/Src/sysmem.c.obj CMakeFiles/dsp-test.elf.dir/Core/Src/system_stm32f4xx.c.obj CMakeFiles/dsp-test.elf.dir/Core/Startup/startup_stm32f411vetx.s.obj -o dsp-test.elf   && cd /home/julian/code/dsp-test/cmake-build-debug && arm-none-eabi-objcopy -Oihex /home/julian/code/dsp-test/cmake-build-debug/dsp-test.elf /home/julian/code/dsp-test/cmake-build-debug/dsp-test.hex && arm-none-eabi-objcopy -Obinary /home/julian/code/dsp-test/cmake-build-debug/dsp-test.elf /home/julian/code/dsp-test/cmake-build-debug/dsp-test.bin
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: warning: dsp-test.elf has a LOAD segment with RWX permissions
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/dsp-test.elf.dir/Core/Src/main.c.obj: in function `SystemClock_Config':
/home/julian/code/dsp-test/Core/Src/main.c:124:(.text.SystemClock_Config+0x56): undefined reference to `HAL_RCC_OscConfig'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /home/julian/code/dsp-test/Core/Src/main.c:138:(.text.SystemClock_Config+0x6c): undefined reference to `HAL_RCC_ClockConfig'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/dsp-test.elf.dir/Core/Src/main.c.obj: in function `main':
/home/julian/code/dsp-test/Core/Src/main.c:73:(.text.main+0x2): undefined reference to `HAL_Init'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/dsp-test.elf.dir/Core/Src/stm32f4xx_it.c.obj: in function `SysTick_Handler':
/home/julian/code/dsp-test/Core/Src/stm32f4xx_it.c:188:(.text.SysTick_Handler+0x2): undefined reference to `HAL_IncTick'
Memory region         Used Size  Region Size  %age Used
             RAM:        1976 B       128 KB      1.51%
           FLASH:        1268 B       512 KB      0.24%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

but i only got that far after deleting .cproject/.mxproject/.project because its seems to not regenerate the CMakeLists at all otherwhise...

Looks to me like a definite problem with STM32CubeMX

christophe0606, 2024-03-06:
I close it since we have concluded it is not directly related to CMSIS-DSP.

closed 2024-03-06 by christophe0606