ARM-software/CMSIS-DSP #160: GCC compiler errors with pendantic flag enabled

felixvanoost, 2024-02-29:
When compiling CMSIS-DSP using GCC 10.3 with the `-pedantic` flag, the compiler produces errors due to the presence of extra semicolons in the `CFFTINIT_` and `RFFTINIT_` macros. One example:

```shell
[build] Source/TransformFunctions/arm_cfft_init_q31.c:140:1: note: in expansion of macro 'CFFTINIT_Q31'
[build]   140 | CFFTINIT_Q31(4096,4096);
[build]       | ^~~~~~~~~~~~
[build] Source/TransformFunctions/arm_cfft_init_q31.c:140:24: error: ISO C does not allow extra ';' outside of a function [-Werror=pedantic]
[build]   140 | CFFTINIT_Q31(4096,4096);
[build]       |                        ^
```

This is happening because the macro definitions all end with semicolons, but an extra semicolon has been added when each macro is called:

```c
#define CFFTINIT_Q31(LEN,LENTWIDDLE)
arm_status arm_cfft_init_##LEN##_q31(
  arm_cfft_instance_q31 * S)
{
    /* Removed for brevity */
};
```

In `arm_cfft_init_q31.c`:

```c
CFFTINIT_Q31(4096,4096);
```

The resulting double semicolons are apparently not allowed in ISO C.

christophe0606, 2024-03-04:
Thanks ! I'll make the correction.

christophe0606, 2024-03-04:
Should be solved in latest commit

closed 2024-03-05 by felixvanoost