ARM-software/CMSIS-DSP #162: Issues keeping tables out of compilation

tvarela7, 2024-03-01:
I'm migrating from older version of CMSIS-DSP, since I need some of the new functions (i.e. `arm_atan2_f32`, `arm_rfft_fast_f32`). Given my current framework, I decided to build libCMSISDSP.a with Makefile, using GCC 11.2 and different flags such as _-mcpu=cortex-m7 -mfpu=fpv5-d16 -Ofast -ffast-math -mfloat-abi=hard_.

I then included it in my main repo, basically replacing the old .a CMSIS-DSP library and include files. I also replaced all fft init functions for the recommended ones (i.e. `arm_cfft_init` now is `arm_cfft_init_512_f32`), as mentioned in readme file to avoid all tables from being included. I'm also using flags such as `-ffunction-sections` and `-fdata-sections` in my final build, for optimization. I've seen that in the past this would be tackled using defines such as `ARM_DSP_CONFIG_TABLES`, but not with the last version.

Moreover, I did not used arm_math.h but rather individually added the needed includes.
i.e.:
```
#include "transform_functions.h"
#include "complex_math_functions.h"
#include "fast_math_functions.h"
```

However, it seems nothing is stopping the final build of overflowing flash memory (like 1mb of unwanted code). I've checked in .map file and all tables are there, even those that I shouldn't be using.

Any ideas of what I might be missing here? maybe some other step other than replacing fft init functions?

Thanks!

christophe0606, 2024-03-04:
@tvarela7 There may be some options to use at link. For `GCC` you need to use `--gc-sections` for linking.

CookiePLMonster, 2024-03-04:
Any particular reason those tables cannot be declared `static`? This explicitly incentivizes the linker to skip them if not referenced.

tvarela7, 2024-03-04:
@christophe0606 thanks for answering. It's solved now. I was actually already using that flag, but I did a silly mistake messing up the Makefiles from CMSIS vs my own project, and I thought I was adding ` -fdata-sections -ffunction-sections` but I was actually just using them in my code...

Now I added `-fdata-sections -ffunction-sections` when compiling CMSIS library and everything seems to be working as expected.

Thanks again!

christophe0606, 2024-03-04:
I am glad to know !

closed 2024-03-06 by christophe0606