ARM-software/CMSIS-DSP #164: arm_sqrt_q31 return value for input value 0

marcluque, 2024-03-04:
When working with the [arm_sqrt_q31](https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FastMathFunctions/arm_sqrt_q31.c) function, I noticed that it returns `ARM_MATH_ARGUMENT_ERROR` for input 0. The documentation specifies that the allowed input range is `[0, +1)`. I would expect 0 to return 0 and give `ARM_MATH_SUCCESS` instead, but it only does the former not the latter. The return value documentation seems to miss the "neutral" case since it specifies:
```
@return        execution status
                   - \ref ARM_MATH_SUCCESS        : input value is positive
                   - \ref ARM_MATH_ARGUMENT_ERROR : input value is negative; *pOut is set to 0
```

Perhaps there's a wrong understanding of the code/documentation on my part.

christophe0606, 2024-03-05:
@marcluque It is indeed a bug. A case is missing in the implementation for 0 argument. Same problem for the `q15` version.

christophe0606, 2024-03-06:
Corrected in commit 0d49aa6db8dabcdc7ee2f16879475a286ab6eb83

marcluque, 2024-03-06:
Great, thank you very much! I'll close this issue then.

closed 2024-03-06 by marcluque