ARM-software/CMSIS-DSP #165: CppCheck giving warning for arm_divide_q31 because of CWE 758 violation

marcluque, 2024-03-07:
I encountered a warning from CppCheck indicating the the following line of [arm_divide_q31.c#L66](https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/FastMathFunctions/arm_divide_q31.c#L66) was violating [CWE 758](https://cwe.mitre.org/data/definitions/758.html).
CppCheck message:
```
cppcheck version="2.5"
msg="Shifting signed 32-bit value by 31 bits is implementation-defined behaviour"
cwe="758"
```

This will also hold for q15 I think.

I suppose that something like `(numerator < 0) ^ (denominator < 0)` could fix it?

christophe0606, 2024-03-11:
Corrected in latest commit

marcluque, 2024-03-11:
Very nice, thanks for the quick turnaround! I'll close this issue then.

closed 2024-03-11 by marcluque