ARM-software/CMSIS-DSP #169: Encountering some issues with the latest DSP version on Keil

zlatanArs, 2024-03-29:
I'm unable to directly translate or summarize content from unseen images or detailed technical descriptions from webpages. However, encountering a large number of errors when trying to compile a project with a specific DSP library version in Keil could be due to several reasons, such as compatibility issues with the library and the microcontroller, incorrect or missing configuration settings, or even the need for additional setup steps that are not automatically handled by selecting the library in the RTE. For specific troubleshooting, it's recommended to refer to the library's documentation, check for known issues with the version in question, and seek advice from the Keil and STM32 communities or support forums.

[image]

christophe0606, 2024-04-02:
@zlatanArs I have just tested on uVision 5.39. I don't see any problem.
Which version are you using ?

What are you doing to get this error window ? Can you list the steps ?

zlatanArs, 2024-04-03:
> @zlatanArs I have just tested on uVision 5.39. I don't see any problem. Which version are you using ?
> 
> What are you doing to get this error window ? Can you list the steps ?

Thank you, I have solved the problem. Previously, I was using MDK, and the compile version in the target was 5. When I changed it to 6 and then compiled it again, there were no problems at all.

closed 2024-04-03 by zlatanArs