ARM-software/CMSIS-DSP #171: Add __WEAK attribute to DSP Functions

renesas-kyle-finch, 2024-04-17:
There are some cases where we need to use our own implementations of some `arm_*` functions. 

If these functions had the `__WEAK` attribute, this would be possible without needing to customize CMSIS-DSP on a case by case basis.

christophe0606, 2024-04-18:
@renesas-kyle-finch May I ask which functions need to be customized ?
What are the needed changes ?

Perhaps those changes may be integrated in the original function.

renesas-kyle-finch, 2024-04-18:
@christophe0606 We have 32-bit multiply accumulator hardware in which we would like to use CMSIS DSP API functions that make use of our hardware registers within some of the DSP functions.

In our case, we have added `__WEAK` to the following functions:
`arm_mult_q31`
`arm_scale_q31`
`arm_biquad_cascade_df1_q31`
`arm_conv_partial_q31`
`arm_conv_q31`
`arm_correlate_q31`
`arm_fir_decimate_q31`
`arm_fir_interpolate_q31`
`arm_fir_q31`
`arm_fir_sparse_q31`
`arm_lms_norm_q31`
`arm_lms_q31`
`arm_mat_mult_q31`
`arm_mat_scale_q31`
`arm_mat_vec_mult_q31`

christophe0606, 2024-04-22:
@renesas-kyle-finch Thanks for feedback. So, indeed in this case __WEAK is the only solution.
I'll plan it for a future release.

christophe0606, 2024-06-28:
Done in commit https://github.com/ARM-software/CMSIS-DSP/commit/69d4b8ed8d3271facb2daff94216f7981774e816

closed 2024-07-09 by christophe0606