ARM-software/CMSIS-DSP #172: Singular Matrix Inversion

thseiler, 2024-04-24:
The current version of CMSIS-DSP arm_mat_inverse_f32() seems to have problems detecting the singularity of certain input matrices:
test_matrix:
 { 1.0, 2.0,
   2.0, 4.0 } 
   
The "flag" that tracks whether a row exchange has happened is not reset at the beginning of the next loop iteration. As soon as a row exchange has happened, the singularity detection logic is short-circuited for the remainder of the processing. A singular matrix that happens to trigger any row exchange, therefore, passes as regular and the status returned by arm_mat_inverse_f32 indicates ARM_MATH_SUCCESS.

Proposed fix:

1) reset the flag at the beginning of each loop:

```
in arm_mat_inverse_f32.c:191
[...]
        /* Loop over the number of columns of the input matrix.
           All the elements in each column are processed by the row operations */
     
        /* Index modifier to navigate through the columns */
        for(column = 0U; column < numCols; column++)
        {
+             /* reset flag */
+             flag = 0; 

              /* Check if the pivot element is zero..
               * If it is zero then interchange the row with non zero row below.
               * If there is no non zero element to replace in the rows below,
               * then the matrix is Singular. */
```

2) Add a singular matrix to the unit tests.

3) Similar situation for _f16, _f32, and _f64

christophe0606, 2024-04-25:
@thseiler Thanks for reporting this.

christophe0606, 2024-06-12:
Corrected in latest commit.

closed 2024-06-24 by christophe0606