ARM-software/CMSIS-DSP #174: Python 3.12 support

elagil, 2024-04-26:
With the release of Ubuntu 24.04 LTS, Python 3.12 is shipped.

Currently, `cmsisdsp` cannot install on it, due to it using the deprecated `distutils`.
The same is true for some dependencies (e.g. numpy 1.22).

I can create a PR, but it will require to update dependencies as well (e.g. to numpy 1.26).

elagil, 2024-04-26:
See also: https://peps.python.org/pep-0632/#migration-advice

christophe0606, 2024-04-29:
@elagil Thanks. I have a PR about Python support that have been pending forever.
I need to find some time to work on the Python support and correct all those problems.

elagil, 2024-04-29:
If you point me to the other PR, I can work on it. We need that support at our company, so I can spend a bit of time on the topic.

Would you be open to move forward to the (somewhat) new way of defining python packages? See https://peps.python.org/pep-0518/

The former `setup.py` or similar will be replaced by a `pyproject.toml` in the process.

christophe0606, 2024-04-29:
@elagil The PR : https://github.com/ARM-software/CMSIS-DSP/pull/148
I just need time to merge and test ... time that I have unfortunately not found so far :-)
I can switch to the new pyproject.toml ... I just need to still be able to support Python 3.9

christophe0606, 2024-06-17:
It should be done in commit https://github.com/ARM-software/CMSIS-DSP/commit/3ebc686ff683cab1ef9b26b076ebe603a10e7437

It was tested only on windows so far.
It has not yet been pushed to PyPI. More tests will be needed.

elagil, 2024-06-17:
@christophe0606 Thanks, I saw that today in the commit history!

I will test on Ubuntu 24.04 and 22.04 and report issues, if any.

closed 2024-07-09 by christophe0606