ARM-software/CMSIS-DSP #175: Overflow Handling in arm_fir_q31 Function

lokesh-0706, 2024-04-29:
In CMSIS/DSP/Source/FilteringFunctions/arm_fir_q31.c, the FIR filter implementation uses wraparound for accumulator overflow and saturation for output overflow. Is this intended behaviour?

And also, Would it be possible to consider providing a more unified approach to overflow handling within this function? Ideally, offering the option to choose either:
- Wraparound for both accumulator and output overflow, or
- Saturation for both accumulator and output overflow.

JonatanAntoni, 2024-04-29:
@lokesh-0706, I moved your question to the new home of CMSIS-DSP and leave it up the maintenance team to respond.

christophe0606, 2024-04-29:
@lokesh-0706 It is intended behavior. With q31 in general CMSIS-DSP requires the signal to be pre-scaled to avoid saturation.
The reason is that instruction set can saturate MAC to q30 but not q31 and for performance reason we do not want to scale to q30, saturate and scale back to q31 for each sample.

Note that if you build for Helium instruction set (Cortex-M55 or M85), you get saturation in this case since it is supported by the vector instruction set (but documentation has not been updated to highlight this).

closed 2024-07-09 by christophe0606