ARM-software/CMSIS-DSP #180: 'WindowFunctions' compiling error

yoc1983, 2024-05-30:
Might be a noob question but I use the library for fft filters etc.. and all is good
now I try to use window functions and I get:
undefined reference to `arm_hamming_f32'

on build log I see all directory but not the 'WindowFunctions'
any idea how to solve this?
seems like it knows about the header but can't find the source file when compiling

christophe0606, 2024-05-30:
@yoc1983  What is the build method you use ? CMSIS build tools ? cmake ? Do you build from the repo or from the CMSIS Pack ?

yoc1983, 2024-05-30:
@christophe0606 Hi, 

I call the functions from a C++ code I write in STM32CubeIDE 
as far as I understand it is using CMake in the background without sharing lot of info or allowing to configure
when I press the build button in the IDE I can see in the log it uses all directory except the 'WindowFunctions'
I can also see the Cmake files in the directory 

I add the CMSIS_DSP through the package manager in the CubeMX I added version 1.15.0
so I guess it is a pack (but I can also see all files and directories like opensource and not a closed library)

also I can see the directory noted in the Makefile :

# Sources
SRCS := $(CMSIS_DSP)/Source/BasicMathFunctions/BasicMathFunctions.c \
 $(CMSIS_DSP)/Source/CommonTables/CommonTables.c \
 $(CMSIS_DSP)/Source/InterpolationFunctions/InterpolationFunctions.c \
 $(CMSIS_DSP)/Source/BayesFunctions/BayesFunctions.c \
 $(CMSIS_DSP)/Source/MatrixFunctions/MatrixFunctions.c \
 $(CMSIS_DSP)/Source/ComplexMathFunctions/ComplexMathFunctions.c \
 $(CMSIS_DSP)/Source/QuaternionMathFunctions/QuaternionMathFunctions.c \
 $(CMSIS_DSP)/Source/ControllerFunctions/ControllerFunctions.c \
 $(CMSIS_DSP)/Source/SVMFunctions/SVMFunctions.c \
 $(CMSIS_DSP)/Source/DistanceFunctions/DistanceFunctions.c \
 $(CMSIS_DSP)/Source/StatisticsFunctions/StatisticsFunctions.c \
 $(CMSIS_DSP)/Source/FastMathFunctions/FastMathFunctions.c \
 $(CMSIS_DSP)/Source/SupportFunctions/SupportFunctions.c \
 $(CMSIS_DSP)/Source/FilteringFunctions/FilteringFunctions.c \
 $(CMSIS_DSP)/Source/TransformFunctions/TransformFunctions.c \
 $(CMSIS_DSP)/Source/WindowFunctions/WindowFunctions.c 

and also in CMakeLists.txt :

###########################
#
# CMSIS DSP
#
###########################

add_library(CMSISDSP STATIC)

include(BasicMathFunctions/Config.cmake)

include(ComplexMathFunctions/Config.cmake)

include(QuaternionMathFunctions/Config.cmake)

include(ControllerFunctions/Config.cmake)
 
include(FastMathFunctions/Config.cmake)
 
include(FilteringFunctions/Config.cmake)
 
include(MatrixFunctions/Config.cmake)

include(StatisticsFunctions/Config.cmake)

include(SupportFunctions/Config.cmake)

include(TransformFunctions/Config.cmake)
 
include(CommonTables/Config.cmake)
  
include(SVMFunctions/Config.cmake)

include(BayesFunctions/Config.cmake)

include(DistanceFunctions/Config.cmake)

include(InterpolationFunctions/Config.cmake)

include(WindowFunctions/Config.cmake)

christophe0606, 2024-05-31:
@yoc1983 I've just looked at the pack description and Windows are not included in version 1.15.0

I see the Windows folder in the pack, but the files are not listed in the pack description file so they won't be built.

Two possible solutions:
1 - You don't use the CMSIS-DSP pack and build from the github version
2 - You use the pack and you copy the Windows folder from the github into your project ..

yoc1983, 2024-05-31:
I understand,
I can see on documentation that it is in the list but OK.

since the installing of the pack wasn't easy, had error and now it is working well I won't touch it.
I will go with option 2 
thanks,

christophe0606, 2024-07-09:
Pack description file has been corrected. New pack is not yet on the pack repository but the ARM.CMSIS.pdsc file is now including the Windows function.

closed 2024-07-17 by christophe0606