ARM-software/CMSIS-DSP #181: Compiling error when using CMSIS-DSP: conflicting types for 'clip_q63_to_q31'; have 'q31_t(q63_t)' {aka 'long int(long long int)'}

effemmeffe, 2024-06-06:
I am trying to calculate the FFT using CMSIS-DSP.
I downloaded the latest CMSIS-DSP from github, copied folders CMSIS\DSP\Include, CMSIS\DSP\PrivateInclude and CMSIS\DSP\Source in my Drivers folder, added CMSIS\DSP\Include and CMSIS\DSP\PrivateInclude to my include path and removed all .c files in CMSIS\DSP\Source that does not have _f32 in the filename, since I am planning to use only float32 data.
My project is build with STM32CubeIDE.

This is the code I'm trying to use:

```
#include "arm_math.h"
#include "arm_const_structs.h"

#define SAMPLES_NUMBER 1024

/* ----------------------------------------------------------------------
* Max magnitude FFT Bin calculation
*
* the input vector must be 1024 elements
* ------------------------------------------------------------------- */

float32_t fft(float32_t *input_vector, uint32_t sampling_rate)
{
	float32_t maxValue;
	uint32_t maxIndex;
	float32_t frequency;
	float32_t output_vector[SAMPLES_NUMBER / 2];
	arm_cfft_instance_f32 fft_handler;
	arm_status status;

	status = ARM_MATH_SUCCESS;

	// Init FFT module
	status = arm_cfft_init_1024_f32(&fft_handler);
	if (status != ARM_MATH_SUCCESS)
	{
		Error_Handler();
	}

	// Process the data through the CFFT/CIFFT module
	arm_cfft_f32(&fft_handler, input_vector, 0, 1);

	// Process the data through the Complex Magnitude Module to calculate the magnitude at each bin
	arm_cmplx_mag_f32(input_vector, output_vector, SAMPLES_NUMBER / 2);

	// Calculates maxValue and returns corresponding BIN value
	arm_max_f32(output_vector, SAMPLES_NUMBER / 2, &maxValue, &maxIndex);

	// Calculates the frequency corresponding to the max BIN value
	frequency = (float32_t)maxIndex * (float32_t)sampling_rate / (float32_t)SAMPLES_NUMBER;

	return frequency;
}
```

When I compile I get the following error:
`conflicting types for 'clip_q63_to_q31'; have 'q31_t(q63_t)' {aka 'long int(long long int)'}`
in file none.h
I noticed that none.h is included in arm_math.h, besides a lot of other include files.
What am I doing wrong?

christophe0606, 2024-06-06:
@effemmeffe What is the compiler you're using ?
I never had this with different compilers like gcc, clang, Arm AC6 or even MSVC ...

I think it may be due to a compile option.

It is not clear what is meant by conflicting type in this context but often it means there are two definitions and the compiler computed different types for both.

But the function is defined as __STATIC_FORCEINLINE so it should no occur unless for some reason __STATIC_FORCEINLINE is not defined in your build.

__STATIC_FORCEINLINE is defined in CMSIS-Core that is used by CMSIS-DSP

`dsp/none.h` is normal. It is to improve portability of library when compiling for different targets.

effemmeffe, 2024-06-06:
I am using arm-none-eabi-gcc that comes with STM32CubeIde.
I have
```
#ifndef   __STATIC_FORCEINLINE                 
  #define __STATIC_FORCEINLINE                   __attribute__((always_inline)) static inline
#endif                                           
```
in cmsis_gcc.h in my path, so it should work.

Can you confirm that I can download the sources from github, setup the correct tree in my project, setup the correct include paths and compile without having to add a library (.a file) to my project?

christophe0606, 2024-06-06:
@effemmeffe It would be good to understand the cause of the problem. But you can also comment the `clip_q63_to_q31` in the header since this function is probably not used in the float versions.

effemmeffe, 2024-06-06:
I found the problem: STM32CubeIde somehow [add here profanity and cursing] removed a bunch of files from the compilation, one of them was arm_math.h so even if I had the include in my code it was not compiled.
Hence the error.
Now I fixed the issue and everything is working.
Thanks, I'll close the issue.