ARM-software/CMSIS-DSP #196: Mismatch on RFFT between Cortx-M4 MCU and Python package

Sixaxis9, 2024-07-13:
Hi,

I am developing a DSP application on a Cortex-M4 based STM32. I am computing the real FFT with arm_rfft_q15.
To make sure my code is correct, I am sending back and forth my data to a python script that runs the cmsis-dsp package available on PyPI.

The issue I am encountering is that the outputs of the MCU code and the Python code sometimes differ by 1 or 2 bits. From my understanding, the output of the two should be exactly the same, and it actually is for other functions like `arm_sub_q15` and `arm_shift_q15`.

Here's a sample input for the function:
```
[   768,    384,   -704,   -256,   -128,    704,   4096,  -2560,
          -1088,  15104,  11840, -13824,  -7936,   2688,  14592,  14784,
           6400,  -3776,  -3904,  -9344, -13312,  -7936,  -3520,  -1216,
          -2432,   -320,   2176,  -1536,   -768,   1024,   2240,   1664,
          -3392,  -1408,  -1088,   -384,    512,   -128,    -64,   -128,
             64,   -192,   -192,     64,    320,    -64,    384,   -192,
              0,      0,      0,      0,      0,      0,      0,      0,
              0,      0,      0,      0,      0,      0,      0,      0],
```

And here's the output of the MCU and Python code respectively (only half since symmetric). As you can see, some elements of the two arrays differ by 1-2 LSB:
```
[ -31,    0,  495,    5,  -15, -732, -803,    3,  189,  788,  553,
       -507, -732, -270, -206,  815,  902,  416,  493, -966, -684, -470,
       -644,  254,  -88,  866,  753,  341,  683, -270,  136, -904, -732,
       -297, -290,  278, -182,  204,   54,  426,  374,  207,  415, -216,
         18, -353, -203, -204, -227,  116,  102,   57,  -78,   64,  163,
         15,  -28, -152,  -94,  -33, -145,  134,   99,  151]
```

```
[ -31,    0,  493,    6,  -15, -732, -803,    5,  188,  789,  553,
       -507, -732, -270, -207,  815,  902,  416,  494, -966, -684, -471,
       -643,  254,  -88,  866,  753,  340,  683, -270,  137, -904, -732,
       -297, -291,  279, -183,  204,   54,  426,  374,  207,  415, -217,
         18, -353, -205, -205, -227,  116,  102,   58,  -78,   64,  163,
         16,  -28, -152,  -94,  -33, -145,  134,   99,  153]
```

Please find below an extract of the code for both platforms:
MCU:
```
volatile int16_t dsp[MAX_SAMPLES] = {0};
int16_t dsp_buffer_0[MAX_FRAMES*2] = {0}; // Real FFT needs 2*N

  arm_rfft_instance_q15 rfft;
  if (arm_rfft_init_q15( & rfft, MAX_SAMPLES, 0, 1) != ARM_MATH_SUCCESS) {
    return -1;
  }

 arm_rfft_q15( & rfft, dsp, dsp_buffer_0);
```
Python:
```
import cmsisdsp as dsp

rfft_inst = dsp.arm_rfft_instance_q15()
dsp.arm_rfft_init_q15(rfft_inst, max_samples, 0, 1)
dsp.arm_rfft_q15(rfft_inst, input_data)
```

Is this expected behaviour for the python package of CMSIS-DSP? Are the two using a different algorithm?
While this is not a cause of major concern in production, I would like to know more about this issue.

Thank you!

christophe0606, 2024-07-15:
@Sixaxis9 There are indeed differences between the implementations for different architectures. It is done for speed reasons and use the instruction set in a natural way.

For RFFT Q15, there are 3 versions:

- A pure C used for M0 and in the Python wrapper.
- A version for M4/M7 using DSP extensions
- A version for M55/M85 using vector instructions (Helium).

Those versions give slightly different results but if I compute the SNR to compare all the versions I am getting (on your example):

```
SNR m0 vs. m7 = 55.74 dB
SNR m0 vs. m55 = 53.63 dB
SNR m7 vs. m0 = 55.73 dB
SNR m7 vs. m55 = 55.73 dB
SNR m55 vs. m0 = 53.63 dB
SNR m55 vs. m7 = 55.73 dB
```

Now if the Q15 version is compared to a double precision reference, the SNR is often below 55 (it depends on the FFT length).

So in practise, the differences between the 3 versions should be too small (compared to other sources of rounding errors) to be a problem.

I agree that it is still a bit annoying that it is not exactly the same result and that you need to revalidate the full chain.

I have used this for the SNR:

```python
def snr(ref,test):
  ref = np.array(ref)
  test = np.array(test)
  signal_energy = np.sum(ref*ref)
  noise_energy = np.sum((ref-test)*(ref-test))
  r = np.where(noise_energy == 0, 0, signal_energy/noise_energy)
  return 10*np.log10(r)
```

To be totally sure, I have compared to the f64 version (converted to Q15):

```python
res = np.round((1<<15)*scipy.fft.rfft(1.0*np.array(input_data)/(1<<15),norm="forward"))

data["f64"] = np.zeros(NB)
data["f64"][0::2] = np.real(res)[:-1]
data["f64"][1::2] = np.imag(res)[:-1]
```

The SNR now gives:

```
SNR m0 vs. f64 = 49.58 dB
SNR m7 vs. f64 = 51.34 dB
SNR m55 vs. f64 = 51.54 dB
SNR f64 vs. m0 = 49.58 dB
SNR f64 vs. m7 = 51.34 dB
SNR f64 vs. m55 = 51.55 dB
```

Sixaxis9, 2024-07-29:
Hi @christophe0606,

Thanks a lot for your very clear and incredibly fast explanation!

I hope this issue will help others who scratch their brain to understand why there are small mismatches between the two implementations :)

Great job with CMSIS-DSP, loving them <3